mpromonet/v4l2rtspserver #3: Nedd for example / more description

borazslo, 2015-03-04:
I would like to ask you some help how to use it well.

- I have a Logitech C920 (it should be able to directly encode in h264) and a new Raspberry Pi
- I have an ubuntu server within the network with nginx rtsp server (or with anything else).
- I would like to send the live stream from the webcam in high quality to the ubuntu server (to reencode and send to ustream/youtube etc. ...)

So I found your repository mentioned [here](http://raspberrypi.stackexchange.com/a/26989). I cloned the repository, built it, installed it without problem. Now how should I use it? First I tried to use with defaults only.
```
	Start V4L2 Capture.../dev/video0
/dev/video0 memory mapping nb buffer:10
/dev/video0 memory mapping buffer:0 size:614400
/dev/video0 memory mapping buffer:1 size:614400
/dev/video0 memory mapping buffer:2 size:614400
/dev/video0 memory mapping buffer:3 size:614400
/dev/video0 memory mapping buffer:4 size:614400
/dev/video0 memory mapping buffer:5 size:614400
/dev/video0 memory mapping buffer:6 size:614400
/dev/video0 memory mapping buffer:7 size:614400
/dev/video0 memory mapping buffer:8 size:614400
/dev/video0 memory mapping buffer:9 size:614400
begin thread
2015-03-04 02:14:29,200 [NOTICE] - /home/borazslo/h264_v4l2_rtspserver/src/main.cpp:64
	Play this stream using the URL "rtsp://192.168.0.100:8554/unicast"
SPS
PPS
profile-level-id=424028;sprop-parameter-sets=Z0JAKLtAUB6/gKJAAAADAEAAAAwYEAALcbAAtx73vheEQjU=,aM44gA==
SPS
PPS
```
_and it is working for ever :)_

But. My VLC in my desktop computer can't open `rtsp://192.168.0.100:8554/unicast`. It says that the connection failed. The `nginx` can't pull this even. Even though `nmap 192.168.0.100 -p 8554` says from Ubuntu that port is open while `h264_v4l2_rtspserver` is running.

So can you help me please, how to use it? Thanks a lot!

borazslo, 2015-03-04:
Well, after installing your v4l2web and some reboot and plug-out/plug-in, theh 264_v4l2_rtspserver began to work and VLC can read it. But I don't know how and why... :( (Is it possible send also the audio?)

mpromonet, 2015-03-04:
From the help (-h option) :
  -W width : V4L2 capture width (default 640)
  -H height: V4L2 capture height (default 480)

You can set capture size using these options. 

When it doesn't works remotely, it is usually helpful to try locally with a simple tool like openRTSP (included in livemedia-utils package or compiling live555)

borazslo, 2015-03-04:
Oh, I'm careless... Thank you. And it works. :)

Is it possible that it's work better with 1920x1080 than with 1280x720? There are moments of "gray-burn-outs" and "pixelizations" with lower resolution. I think that 1920x1080 works better because it is its native format so there is no need to work on. Is it true?

(Audio would be good but not so important. )

mpromonet, 2015-03-23:
SPS and PPS are information on the H264 stream.
This is parsed by the program because depending on the source this can appears only once or be repeated (it seems your case)
Following your suggestion I removed this log from the default log level.

closed 2015-03-25 by mpromonet