mpromonet/v4l2rtspserver #4: Timeout after 25 minute

trustn0one, 2015-03-22:
Hi. 
I have a timeout after about 25 minutes. I can't find the cause. 
After this, if i try to resend the command to start the process again, i see this on prompt:

015-03-22 15:39:57,498 [NOTICE] - /h264_v4l2_rtspserver/src/main.cpp:240
	Create V4L2 Source.../dev/video0
2015-03-22 15:39:57,917 [NOTICE] - /h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:94
	driver:raspicam 1000001
2015-03-22 15:39:57,919 [NOTICE] - /h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:102
	/dev/video0 support read/write
2015-03-22 15:39:57,921 [ERROR ] - /h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:108
	Mandatory capability not available for device:/dev/video0
2015-03-22 15:39:58,272 [ERROR ] - /h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:43
	Cannot init device:/dev/video0

I'm not able to start process anymore even if i try to restart the RaspPi.
Thanks in advance.

trustn0one, 2015-03-22:
Retyping again  :
sudo modprobe -v bcm2835-v4l2
the cam starts (don't know why)
However resending the command with verbose mode, this is the output until the timeout occur (10 min this time)

2015-03-22 15:53:05,001 [INFO  ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1427039585 fps:25 bandwidth:4867kbps

2015-03-22 15:53:05,003 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.1675	size:24516	diff15ms	queue:0
2015-03-22 15:53:05,026 [INFO  ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1427039585 fps:25 bandwidth:4866kbps

2015-03-22 15:53:05,028 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.26314	size:23886	diff0ms	queue:0
2015-03-22 15:53:05,033 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.26314	size:23886	diff2ms	queue:1
2015-03-22 15:53:05,041 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.41695	size:23882	diff15ms	queue:0
2015-03-22 15:53:05,068 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.68387	size:23930	diff0ms	queue:0
2015-03-22 15:53:05,073 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.68387	size:23930	diff1ms	queue:1
2015-03-22 15:53:05,075 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.75044	size:23926	diff6ms	queue:0
2015-03-22 15:53:05,106 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.106271	size:24051	diff0ms	queue:0
2015-03-22 15:53:05,111 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.106271	size:24051	diff1ms	queue:1
2015-03-22 15:53:05,121 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.121618	size:24047	diff15ms	queue:0
2015-03-22 15:53:05,146 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.146472	size:24139	diff0ms	queue:0
2015-03-22 15:53:05,152 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.146472	size:24139	diff1ms	queue:1
2015-03-22 15:53:05,154 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.154642	size:24135	diff8ms	queue:0
2015-03-22 15:53:05,186 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.186237	size:24080	diff0ms	queue:0
2015-03-22 15:53:05,191 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.186237	size:24080	diff1ms	queue:1
2015-03-22 15:53:05,201 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.201636	size:24076	diff15ms	queue:0
2015-03-22 15:53:05,226 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.226171	size:23876	diff0ms	queue:0
2015-03-22 15:53:05,231 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.226171	size:23876	diff1ms	queue:1
2015-03-22 15:53:05,235 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.235036	size:23872	diff8ms	queue:0
2015-03-22 15:53:05,268 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.268240	size:23922	diff0ms	queue:0
2015-03-22 15:53:05,273 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.268240	size:23922	diff1ms	queue:1
2015-03-22 15:53:05,275 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.274870	size:23918	diff6ms	queue:0
2015-03-22 15:53:05,306 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.306218	size:24276	diff0ms	queue:0
2015-03-22 15:53:05,312 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.306218	size:24276	diff1ms	queue:1
2015-03-22 15:53:05,321 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.321645	size:24272	diff15ms	queue:0
2015-03-22 15:53:05,346 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.346164	size:24154	diff0ms	queue:0
2015-03-22 15:53:05,351 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
	queueFrame	timestamp:1427039585.346164	size:24154	diff1ms	queue:1
2015-03-22 15:53:05,354 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.354637	size:24150	diff8ms	queue:0
2015-03-22 15:53:05,386 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.386356	size:23950	diff0ms	queue:0
2015-03-22 15:53:05,392 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:228
[... 54 lines cut ...]
	queueFrame	timestamp:1427039585.745903	size:24242	diff1ms	queue:1
2015-03-22 15:53:05,754 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.754625	size:24238	diff8ms	queue:0
2015-03-22 15:53:05,786 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.785908	size:24107	diff0ms	queue:0
[... cut ...]
	queueFrame	timestamp:1427039585.868767	size:23918	diff1ms	queue:1
2015-03-22 15:53:05,875 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.875757	size:23914	diff6ms	queue:0
2015-03-22 15:53:05,906 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:205
	getNextFrame	timestamp:1427039585.905904	size:24489	diff0ms	queue:0
2015-03-22 15:53:05,911 [DEBUG ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:169
	deliverFrame	timestamp:1427039585.911399	size:24485	diff5ms	queue:0

mpromonet, 2015-03-22:
Is there any log in /var/log/messages concerning the bcm2835-v4l2 module ?

trustn0one, 2015-03-22:
Thank for your. With a grep command i obteined the following:

/var/log/debug:Mar 22 15:49:23 raspberrypi kernel: [  110.608137]
bcm2835-v4l2: scene mode selected 0, was 0
/var/log/syslog:Mar 22 15:49:23 raspberrypi kernel: [  110.608137]
bcm2835-v4l2: scene mode selected 0, was 0
/var/log/syslog:Mar 22 15:49:23 raspberrypi kernel: [  110.616151]
bcm2835-v4l2: V4L2 device registered as video0 - stills mode >
1280x720
/var/log/syslog:Mar 22 15:49:23 raspberrypi kernel: [  110.621733]
bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
/var/log/messages.1:Mar 21 15:17:56 raspberrypi kernel:
[163010.642911] bcm2835-v4l2: V4L2 device registered as video0 -
stills mode > 1280x720
/var/log/messages.1:Mar 21 15:17:56 raspberrypi kernel:
[163010.648370] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver
0.0.2 loaded.
/var/log/debug.1:Mar 21 15:17:56 raspberrypi kernel: [163010.637585]
bcm2835-v4l2: scene mode selected 0, was 0
/var/log/auth.log.1:Mar 21 15:17:56 raspberrypi sudo:       pi :
TTY=pts/0 ; PWD=/dev ; USER=root ; COMMAND=/sbin/modprobe bcm2835-v4l2
/var/log/auth.log.1:Mar 21 15:42:57 raspberrypi sudo:       pi :
TTY=pts/0 ; PWD=/h264_v4l2_rtspserver ; USER=root ;
COMMAND=/sbin/modprobe -v bcm2835-v4l2
/var/log/kern.log:Mar 22 15:49:23 raspberrypi kernel: [  110.608137]
bcm2835-v4l2: scene mode selected 0, was 0
/var/log/kern.log:Mar 22 15:49:23 raspberrypi kernel: [  110.616151]
bcm2835-v4l2: V4L2 device registered as video0 - stills mode >
1280x720
/var/log/kern.log:Mar 22 15:49:23 raspberrypi kernel: [  110.621733]
bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
/var/log/messages:Mar 22 15:49:23 raspberrypi kernel: [  110.616151]
bcm2835-v4l2: V4L2 device registered as video0 - stills mode >
1280x720
/var/log/messages:Mar 22 15:49:23 raspberrypi kernel: [  110.621733]
bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
/var/log/auth.log:Mar 22 15:49:23 raspberrypi sudo:       pi :
TTY=pts/0 ; PWD=/ ; USER=root ; COMMAND=/sbin/modprobe -v bcm2835-v4l2
/var/log/auth.log:Mar 22 21:23:19 raspberrypi sudo:       pi :
TTY=pts/0 ; PWD=/ ; USER=root ; COMMAND=/bin/grep -r bcm2835-v4l2
/var/log/
/var/log/auth.log:Mar 22 21:23:51 raspberrypi sudo:       pi :
TTY=pts/0 ; PWD=/ ; USER=root ; COMMAND=/bin/grep -r bcm2835-v4l2
/var/log/
/var/log/kern.log.1:Mar 21 15:17:56 raspberrypi kernel:
[163010.637585] bcm2835-v4l2: scene mode selected 0, was 0
/var/log/kern.log.1:Mar 21 15:17:56 raspberrypi kernel:
[163010.642911] bcm2835-v4l2: V4L2 device registered as video0 -
stills mode > 1280x720
/var/log/kern.log.1:Mar 21 15:17:56 raspberrypi kernel:
[163010.648370] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver
0.0.2 loaded.
/var/log/syslog.1:Mar 21 15:17:56 raspberrypi kernel: [163010.637585]
bcm2835-v4l2: scene mode selected 0, was 0
/var/log/syslog.1:Mar 21 15:17:56 raspberrypi kernel: [163010.642911]
bcm2835-v4l2: V4L2 device registered as video0 - stills mode >
1280x720
/var/log/syslog.1:Mar 21 15:17:56 raspberrypi kernel: [163010.648370]
bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.

I hope this will be usefull form you.

mpromonet, 2015-03-23:
Did you get same issue with the user space V4L2 driver http://www.linux-projects.org/modules/sections/index.php?op=viewarticle&artid=14 ?

trustn0one, 2015-03-23:
Sorry,  what issue are you talking about? the page you suggest, shows how to install V4L2...or is there something i can't get?

Thanks

trustn0one, 2015-03-23:
I think i figure it out. 
I changed the acquiring resolution paramenters from 1280 to 1024. It is running by 2 hours with no disconnections. Are there any constrain concerting this?

mpromonet, 2015-03-23:
The code of this project should not have resolution limitation, but the V4L2 drivers (bcm2835-v4l2 or uv4l) could have some limitations.

trustn0one, 2015-03-27:
Just to let you know that solution (small resolution) doens't work. I've made a brand new installation of raspbian with just the component and dipendence your library needs.
I'll advice you if something strange will happen.
Just a question, if i have to exit from the task i press CTRL+C, and this is what i obtein from bash, is it normal?

^CSIGINT
2015-03-27 15:52:32,388 [NOTICE] - /h264_v4l2_rtspserver/src/main.cpp:273
	Exiting....
libv4l2: error dequeuing buf: Invalid argument
VIDIOC_DQBUF: Invalid argument
2015-03-27 15:52:32,486 [NOTICE] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:190
	V4L2DeviceSource::getNextFrame errno:22 Invalid argument

2015-03-27 15:52:32,488 [ERROR ] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:107
	error:Invalid argument

2015-03-27 15:52:32,489 [NOTICE] - /h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:118
	end thread

trustn0one, 2015-03-27:
Nothing to so, it stop!!
But i got this error now:

 glibc detected *** h264_v4l2_rtspserver/h264_v4l2_rtspserver: double free or corruption (!prev): 0xb592e1a8 ***

This could help you?

mpromonet, 2015-03-29:
To identify the problem a gdb backtrace will be better.
I fix a memory corruption with commit https://github.com/mpromonet/h264_v4l2_rtspserver/commit/ff30f1b8a862bd61f0c1d27c491d849c8f080c0f.
Is it still crash ?

trustn0one, 2015-03-29:
thanks Michel, i'll let you know...

trustn0one, 2015-03-29:
It stop again, but now the error is:

"Segmentation fault"

I've never used gdb backtrace. If you can suggest me how to use it...
thanks.

mpromonet, 2015-03-29:
I let it run an hour without crash, with 2 clients... I will try to longer run with bigger resolution.
The log V4L2DeviceSource::doStopGettingFrames indicate a client close a connection.
In order to get a core, you need to allow to create it usually the limit for core is 0. You could do this running :

    ulimit -c unlimited && h264_v4l2_rtspserver
When it stop you should see something like : 

    Segmentation fault      (core dumped) h264_v4l2_rtspserver
Next you can get backtrace for all thread with : 

    gdb h264_v4l2_rtspserver ./core -ex 'thread apply all where' -ex quit

It could be helpful if you can post it.

trustn0one, 2015-03-30:
Hope this could help u

[New LWP 2200]
[New LWP 2199]
Core was generated by `h264_v4l2_rtspserver/h264_v4l2_rtspserver -F 25 -W 1280 -H 720 -P 8555 /dev/vid'.
Program terminated with signal 11, Segmentation fault.
#0  0x0002ab6c in ?? ()

Thread 2 (LWP 2199):
#0  0xb6d02d7c in ?? ()
#1  0xb6d16eb8 in ?? ()
#2  0xb6d16eb8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (LWP 2200):
#0  0x0002ab6c in ?? ()
#1  0x0002a728 in ?? ()
#2  0x0002a728 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

mpromonet, 2015-03-30:
Not really, it just says that both thread stack are smashed... 
I will make some tests with 1280x720 resolution.

trustn0one, 2015-03-30:
ok, just to be precise..
i've started the stream at 10.00 am and until 6.00 pm all work well. Then i've made CTRL+c to stop process, then restart...and after 30 min it stopped again with "segmentation fault" error message

mpromonet, 2015-03-30:
There were big mistakes between reader/consumer threads.
I fixed this protecting the queue with a mutex in commit https://github.com/mpromonet/h264_v4l2_rtspserver/commit/99b4c42a07ea32d7dbbc950c473d9a48abde1541
It should be better now.
It is possible to disable the thread using -s parameter but this impact framerate.

trustn0one, 2015-03-31:
Hi, 
it is running from 2 hours with no issue. Just a thing..i have to reduce the framerate from 25 to 20 because the client can't  reprocude the video stream.
I let you know in case of other problems.

Thanks.

trustn0one, 2015-03-31:
Update: no more segmentation fault happens.
But...i notice this: the bitrate is unstable -> during streaming on client  i see values around 120 Kb/s, then suddenly go to 0 Kb/s, then return around 120 Kb/s (i used 25, 20, 15 FPS)
This is from your last update. 
May be it depens becouse the buffer queue is empty so the process has to wait to refill it?

Thanks.

trustn0one, 2015-03-31:
Update 2: still no fault after 9 hours! Good Job!
But remain the 0 Kb/s bitrate for just a while. I tried to record the stream, and it crops just some frame, but it's acceptable. Do you know the possible cause?

mpromonet, 2015-04-01:
Thanks for pointing out this issue.

About bitrate I am surprise because a video 1280x720 25fps is probably above 1Mbits/s. 
I did not reproduce the problem you described. It could be due to udp buffer overflow.
Anyway it is a different subject, so it is better to open an other issue

closed 2015-04-01 by mpromonet