mpromonet/v4l2rtspserver #5: How can I make multiple video streaming on RTSP server

michael5430, 2015-04-08:
Hello Michel Promonet:

I'm Michael comes from Taiwan. First, thanks for your contribution on this amazing code.
I'm interested in your RTSP project and I would like to know how can I make multiple video streaming on this project. I just bought two Logitech c920 webcam and I also complied your code on Raspberry Pi B+. It worked perfectly. But when I traced the code, I had little confuse about how to make two video streaming on two webcam. It means I can select witch webcam I want to watch instantly using VLC player. Hope you can give me some tips. Thanks you very much.

Michael

mpromonet, 2015-04-09:
Hi Michael,

By now the code manage only one V4L2 device and listen for incoming RTSP requests, then you will need to run one program per camera.

You could do using 2 different RTSP port : 

    h264_v4l2_rtspserver -P 8554 /dev/video0 
    h264_v4l2_rtspserver -P 9554 /dev/video1
Then you could access to the first camera with rtsp://<raspberrypi>:8554/unicast and the second one using rtsp://<raspberrypi>:9554/unicast

Otherwise it is not really complex to modify the code in order to create severals V4L2DeviceSource and append a UnicastServerMediaSubsession to the RTSP server.

Michel

michael5430, 2015-04-10:
Hi Michel,

Thanks for your reply. As you said if I want to add another camera in the code, should I do follow step below?
1. Create V4L2DeviceSource
2. Create StreamReplicator and put V4L2DeviceSource in it
3. Append a UnicastServerMediaSubsession to the RTSP server
Is that right?

I have a little confuse about your newest update, you add thread lock in V4L2DeviceSource::queueFrame and V4L2DeviceSource::deliverFrame, Does other code use the queue concurrently?

Michael

mpromonet, 2015-04-12:
Hi Michael,

The steps you describe are right. It is easy to modify the code. The main problem is to pass parameters for width,height,uri,... or use a configuration file.

Concerning mutex, it is really needed because the queue is filled in the reader thread and consumed  by the live555 mainloop (see issue #4 ). I will extract the queue in a separated class to improve readability.

Best Regards,

Michel.

michael5430, 2015-04-15:
Hi Michel,

Really thanks for your help. Now I know how to modify it.

Michael

mpromonet, 2015-08-29:
I include this feature in this commit https://github.com/mpromonet/h264_v4l2_rtspserver/commit/49a978595a5a9d9f12ca1c1c92b7d636b75b51ee

closed 2015-08-29 by mpromonet