mpromonet/v4l2rtspserver #6: Quality problem

borazslo, 2015-05-15:
I used h264_v4l2_rtspserver with success. But now the output image is not good any more (half gray, blinking, etc.) :( Can you help me?

- Background
     - Raspberry Pi + Logitech C920 webcam
     - fresh h264_v4l2_rtspserver, and sudo apt-get update-upgrade
- my command: `./h264_v4l2_rtspserver -W 1920 -H 1080 -F 15 -u minotsledge -P 1985 -v`
- the image with VLC on local network: [image]
- from the log:
```
2015-05-15 09:23:24,789 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/inc/logger.h:44
	level:INFO
2015-05-15 09:23:24,790 [NOTICE] - ./h264_v4l2_rtspserver/src/main.cpp:249
	Create V4L2 Source.../dev/video0
2015-05-15 09:23:24,793 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:94
	driver:uvcvideo 85000001
2015-05-15 09:23:24,793 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:102
	/dev/video0 support read/write
2015-05-15 09:23:24,793 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:103
	/dev/video0 support streaming
2015-05-15 09:23:24,795 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:144
	bufferSize:4147200
2015-05-15 09:23:24,796 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:163
	fps:1/15
2015-05-15 09:23:24,797 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:164
	nbBuffer:0
2015-05-15 09:23:24,797 [NOTICE] - ./h264_v4l2_rtspserver/src/main.cpp:255
	Start V4L2 Capture.../dev/video0
2015-05-15 09:23:24,849 [NOTICE] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:66
	Device /dev/video0 nb buffer:10
2015-05-15 09:23:24,850 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:0 size:4147200
2015-05-15 09:23:24,852 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:1 size:4147200
2015-05-15 09:23:24,855 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:2 size:4147200
2015-05-15 09:23:24,857 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:3 size:4147200
2015-05-15 09:23:24,859 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:4 size:4147200
2015-05-15 09:23:24,860 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:5 size:4147200
2015-05-15 09:23:24,862 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:6 size:4147200
2015-05-15 09:23:24,863 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:7 size:4147200
2015-05-15 09:23:24,865 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:8 size:4147200
2015-05-15 09:23:24,866 [INFO  ] - ./h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapCapture.cpp:85
	Device /dev/video0 buffer idx:9 size:4147200
2015-05-15 09:23:25,027 [NOTICE] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:98
	begin thread
2015-05-15 09:23:25,027 [NOTICE] - ./h264_v4l2_rtspserver/src/main.cpp:62
	Play this stream using the URL "rtsp://192.168.0.100:1985/minotsledge"
2015-05-15 09:23:25,182 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696205 fps:1 bandwidth:94kbps
2015-05-15 09:23:25,183 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:57
	SPS size:38
2015-05-15 09:23:25,183 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:58
	PPS size:4
2015-05-15 09:23:25,183 [NOTICE] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:84
	profile-level-id=424028;sprop-parameter-sets=Z0JAKLtAPAES8uAokAAAAwAQAAADAeYEAALcbAAtx73vheEQjUA=,aM44gA==
2015-05-15 09:23:25,183 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
	IDR size:2948
2015-05-15 09:23:25,184 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
	IDR size:2957
2015-05-15 09:23:25,184 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
	IDR size:3007
2015-05-15 09:23:25,184 [INFO  ] - ./h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
	IDR size:3131
2015-05-15 09:23:26,051 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
[... 728 lines cut ...]
	intv_sec:1431696358 fps:15 bandwidth:2926kbps
2015-05-15 09:25:58,065 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696358 fps:64 bandwidth:3131kbps
2015-05-15 09:25:59,058 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696359 fps:15 bandwidth:2935kbps
2015-05-15 09:25:59,069 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696359 fps:60 bandwidth:2918kbps
2015-05-15 09:26:00,001 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696360 fps:56 bandwidth:2724kbps
2015-05-15 09:26:00,058 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696360 fps:15 bandwidth:2922kbps
2015-05-15 09:26:01,058 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696361 fps:15 bandwidth:2919kbps
2015-05-15 09:26:01,065 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696361 fps:64 bandwidth:3125kbps
2015-05-15 09:26:02,058 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696362 fps:15 bandwidth:2935kbps
2015-05-15 09:26:02,068 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696362 fps:60 bandwidth:2925kbps
2015-05-15 09:26:03,000 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	outtv_sec:1431696363 fps:56 bandwidth:2739kbps
2015-05-15 09:26:03,058 [INFO  ] - ./h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:33
	intv_sec:1431696363 fps:15 bandwidth:2934kbps

```

mpromonet, 2015-05-17:
Commonly this occurs when network (including ip stacks) loose I-frames. 
What is strange is that the capture is about 3Mbits/s, then not so big.
You should try to use RTP over TCP (option --rtsp-tcp of VLC) to see if something change.
If the network is not good enough (or the IP stack to small) you should have a better video with bigger latencies.

ronnyvdbr, 2015-08-21:
Hi, you did great work on this, it's really usefull, I use it as camera streamer on a Raspberry Pi.
I have a question though.
I can configure the UV4L2 driver to jpeg stills at 15 ftps on a resolution of 2592x1944 px, but ...
I have noticed if I start your rtspserver, that the uv4l2 driver is automatically set back to H264 mode, and the resolution switches back to 1920x1080.
Is it possible to circumvent the setback?

mpromonet, 2015-08-21:
Hi Ronny,
It could be nice to start a new discussion, better than continue this one that is ralated to something else.
I am not sure to understand what do you expect, this rtspserver try to set format(H264), size and fps depending on its arguments (-H, -W and -F).
Are you expecting that it to restore format/resolution after it stops ? or avoid to set format/resolution and use the current configuration ?
Let me know.
Regards,
Michel.

wh279813, 2015-10-24:
Hi Michel,
I have the same QoS problem on RaspberryPi, but it performs better on my Laptop(Ubuntu).
Also in the log given by borazslo, bandwidth is limited in 3Mbits/s. 
The same happens to me, why it can not break though 3Mbits/s? Local network issue?

mpromonet, 2015-10-24:
Hi HuAn,
For capturing 1920x1080 15fps, 3Mbits/s should be enough, how much are you expecting ?

The capture bandwidth depend also of encoder configuration that you can get using `v4l2-ctl --all` : 

    Codec Controls

             video_bitrate_mode (menu)   : min=0 max=1 default=0 value=0 flags=update
                  video_bitrate (int)    : min=25000 max=25000000 step=25000 default=10000000 value=10000000
         repeat_sequence_header (bool)   : default=0 value=0
            h264_i_frame_period (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level (menu)   : min=0 max=11 default=11 value=11
                   h264_profile (menu)   : min=0 max=4 default=4 value=4

Regards,
Michel.

wh279813, 2015-10-26:
Hi Michel,
Thanks for you replay.
 3Mbits/s is indeed enough for  1920x1080 15fps or 1270x920 30fps. Last few days I have found the problem is on the VLC player. Even for local resource, video still have quality problem. It seems that VLC cannot support H264 decoding perfectly. And I am looking for an alternative over VLC.
Best Regards
HuAn

closed 2015-11-24 by mpromonet