mpromonet/v4l2rtspserver #7: Cannot set format format

robertxee, 2015-06-18:
Hi,

I'm getting this following error:

pi@raspberrypi ~/h264_v4l2_rtspserver $ ./h264_v4l2_rtspserver -vv /dev/video0
2015-06-18 15:26:28,313 [INFO  ] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/inc/logger.h:44
        level:DEBUG
2015-06-18 15:26:28,314 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:249
        Create V4L2 Source.../dev/video0
2015-06-18 15:26:28,317 [NOTICE] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:94
        driver:usbtv 85200001
2015-06-18 15:26:28,317 [NOTICE] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:102
        /dev/video0 support read/write
2015-06-18 15:26:28,317 [NOTICE] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:103
        /dev/video0 support streaming
2015-06-18 15:26:28,318 [ERROR ] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:133
        Cannot set format format:1448695129
2015-06-18 15:26:28,318 [ERROR ] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Capture.cpp:43
        Cannot init device:/dev/video0
pi@raspberrypi ~/h264_v4l2_rtspserver $

Could you please assist me?

Thanks,
Robert

mpromonet, 2015-06-18:
Hi Robert,

I guess your device simply does not support H264 capture.
You could check this with the command : 

             v4l2-ctl --list-formats

I will put a better log, it indicates that the format is YUYV instead of H264. 
You should consider that this code does not compress, it read an H264 capture. 

Regards
Michel.

wh279813, 2015-08-09:
Hi,mpromonet
Could you recommend a webcam which support H264 capture in 640*480 definition?
As I know for most Logitech webcams, h264 compression only works in 720p and 1080p.

Best Regards
Harry

mpromonet, 2015-09-05:
Hi Robert,
If you would like to compress the camera capture in H264 in order to feed the h264_v4l2_rtspserver you could use a intermediate virtual device [v4l2loopback](https://github.com/umlaeute/v4l2loopback) and v4l2compress_h264 (from [v4l2tools](https://github.com/mpromonet/v4l2tools)) to compress the video stream.

If your camera device is  /dev/video0 you should be able to use these commands : 

     modprobe v4l2loopback video_nr=10
     v4l2compress_h264 /dev/video0 /dev/video10 &
     h264_v4l2_rtspserver /dev/video10 &

These commands will : 
 - create a virtual video device /dev/video10
 - read YUYV frame from /dev/video0, compress them using libx264, write H264 frames to /dev/video10
 - read H264 frames from /dev/video10 to feed the RTSP streamer.

Best Regards,
Michel.

closed 2015-10-17 by mpromonet