mpromonet/v4l2rtspserver #9: Switch beween streaming mode - possible?

ronnyvdbr, 2015-08-21:
Dear Michel, sorry for the other thread, starting a new one hereby, hope U can help me out.
The uv4l2 driver from Linux projects supports a few different streaming modes, namely:
H264
Mjpeg
Jpeg stills
YUV4.2.0

The uv4l2 driver can be configured to use jpeg stills format at 15 fps at 2592x1944 pixels.
When I configure this,start the driver, and subsequently start your rtsp server, then the driver reverts to using format h264, and by this, falling back to resolution 1920x1088.
I would like to know if it would be possible to alter the behaviour of the rtsp server so it would accept the configured jpeg stills format at 15 fps on 2592x1944 pixels.

I already checked the live555 library documentation, and I am not sure if it would take that input format, but maybe U can tell me otherwise.

mpromonet, 2015-08-21:
Hi Ronny,
No problem, just clearer to not mix to much the threads...
This server use H264 capture, in order to feed the RTSP server with JPEG capture live555 miss some code (see http://www.live555.com/liveMedia/faq.html#jpeg-streaming).
I had submited some years ago a partial solution to brings video/JPEG support to live555 RTSP server (see http://lists.live555.com/pipermail/live-devel/2012-February/014672.html). It is not so complex to substitute the H264 VideoSource with a MJPEG VideoSource.
However I guess 2592x1944 @15fs JPEG will be a bit too much for a raspberry ?
Regards,
Michel.

ronnyvdbr, 2015-08-25:
Hi, thanks for the response.
I have read through your articles and I find it regrettable that nobody provided U that time with the necessary information to complete your coding.
Although jpeg is indeed not the most interesting compression solution for streaming since each frame is a key frame, it would have been a nice addition to the live555 libraries for people having dedicated bandwith on local networks.
Unfortunately I lack sufficient coding knowledge to do it myself, but trying to digest the JPEG rfc made me dazzle.
I noticed however that the articles report an existing solution from some camera product "ElphelJPEGDeviceSource", going through the documentation it should be able to use a jpeg file sink and contains also an rtsp server, would it be usable in this case?  Compiling this together would be doable for me, however, the setting up the cross compile environment article seems to be missing.
Normally 2592x1944@15 fps should be possible for the raspberry since it contains a dedicated gpu for this purpose, the image sensor that they sell with the raspberry is capable of doing this frame rate in jpeg stills.

closed 2015-08-30 by ronnyvdbr