mpromonet/v4l2rtspserver #10: Log file configuration

ronnyvdbr, 2015-08-29:
Hi, I'm currently working out a raspberry configuration with a read only file system.
The problem is that the rtspserver is writing a log file to /var/log which is not writeable.
Because of that the server won't start.

Is it in any way possible to configure the location of the log file?

mpromonet, 2015-08-29:
Hi Ronny,
The h264_v4L2_rtspserver write to standard output using log4cpp.
If you get log in /var/log, I guess you redirect to syslog or whatever starting it, isn't it ?
Regards,
Michel.

ronnyvdbr, 2015-08-30:
Hi Michel, thanks for your reaction, indeed it was redirected with the init script which was starting the server.
Case closed.

closed 2015-08-30 by ronnyvdbr