mpromonet/v4l2rtspserver #13: audio support

3lixy, 2015-11-06:
Hi Michel,

Thanks for your work on this RTSP server it is very stable and the only stable one I have found for the pi. Great work :D.

Are there any plans to add audio support?

I had a look at the Live555 docs, but alas I have not had to deal with C and the more I read the more my head hurt.

I was thinking of something like this: http://www.ebay.co.uk/itm/271382734793 as its small and would work with the raspberry pi.

When plugged into a pi running raspbian jessie:

[   53.047544] usb 1-1.2: new full-speed USB device number 4 using dwc_otg
[   53.154968] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=013c
[   53.155006] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   53.155026] usb 1-1.2: Product: USB PnP Sound Device
[   53.155043] usb 1-1.2: Manufacturer: C-Media Electronics Inc.
[   53.182824] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:0D8C:013C.0001/input/input0
[   53.238085] hid-generic 0003:0D8C:013C.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-20980000.usb-1.2/input2
[   53.462858] usbcore: registered new interface driver snd-usb-audio

Thanks again for your work on this.

marko3dana, 2015-12-30:
Is there any news on audio ?
It would be really good feature.... and I would also use it...
However thank you for great work

mpromonet, 2016-01-09:
Hi marko3dana,
Feel free to submit a pull request.
Regards,
Michel.

mpromonet, 2016-10-09:
Hi,
Recently I added a first support of ALSA capture using `-A <alsa capture device>`, it add an RTSP stream `/audio`.
I will probably rework it in order to use an RTSP subsession.
Regards,
Michel.

ronnyvdbr, 2016-11-18:
Hi Michel, this sounds awesome, would be a very nice addition to the rtsp server.  Currently I don't have an idea how a vcr would access this stream, but as u already say it's possibly via a subsession in your rtsp stream.  I have some experience with gstreamer framework, but i guess it's implementation is entirely different than live 555.  I guess u need to mux the video and audio stream and the connecting end will demux it somehow.

mpromonet, 2016-12-18:
Hi Ronny,
There is different way to do this, muxing as a TransportStream, or using 2 subsessions of the same RTSP session.
You can find a recent try to do this in this [commit](https://github.com/mpromonet/v4l2rtspserver/commit/877eab616dfe2fd17fccdbda17eed37eecd8400b).
Best Regards,
Michel.

scramble45, 2017-02-06:
How do you specify a ALSA device? I changed my default ALSA card to my capture device but I am still getting: 
````
2017-02-06 22:02:15,466 [NOTICE] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
        profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQCgC3QDxImo=,KO4CXLA=
cannot set sample format (Invalid argument)
````
I am passing:
````
 -A 22050
````

mpromonet, 2017-02-06:
Hi scramble45,

You can specify the alsa device using : /dev/video0,<alsaDevice>.
But the error setting the format occurs executing : 

> snd_pcm_hw_params_set_format (m_pcm, hw_params, SND_PCM_FORMAT_S16_BE)

Maybe your card doesnot support it, you can try to change to an other format that could be transport throught RTP.

Best Regards,
Michel.

scramble45, 2017-02-07:
So for example: -I /dev/video0,hw:1,0 ?

Thanks

foreverYoungGitHub, 2017-04-17:
Hi, mpromonet,

Thank you for your great implementation.

Recently, I want to create the video with audio streaming. However, when I try to specify the audio device, it can not begin a stream. I was wondering how to specify the input of the audio.

The audio device is a part of webcam. It can not be found in /dev/audio* and could not open by default. I try to find it in /dev/snd/. There is a pcm sound, while it not works to create stream. I also try to list the by-path folder such as ls -il by-path, it still not work. When I use the hw:1,0 which can be capture by vlc, but it also can not create the stream.

Could you give me some hint to specify the parameter of the folder of the input audio?

Thanks a lot,

Best wishes,

Yang Liu

mpromonet, 2017-04-19:
Hi Yang Liu,
You can get the list of ALSA capture device using:

     arecord -l

Then you should be able to capture one using :

     ,(alsa decice)

Best Regards
Michel

closed 2017-05-13 by mpromonet