mpromonet/v4l2rtspserver #14: Capture image while serving

nicogarcia, 2015-12-02:
Hello,

First of all, I want to thank you very much for your code, it's wonderful!

This issue is to ask you for some help, I'm trying to capture an image from the camera in a Raspi while streaming and I get 'Device busy' message (Currently I'm using picamera python library)

So I read about v4l2loopback for making virtual devices and installed successfully in my Rpi.

Now I would like to know how to make /dev/video0 (camera) being piped to /dev/video1 and /dev/video2 to be able to capture image from video1 and streaming from video2.

Is this approach right and possible? How can I do that? (I have gstreamer-1.0 installed)

Thank you very much for your help!

mpromonet, 2015-12-02:
Hi Nicolas,
Your approach is right, you can open several times the virtual device of v4l2loopback. So if you don't need to transcode, you can capture the camera (/dev/video0) and write to a virtual device (/dev/video1) and open twice /dev/video0.

You can look to a small program v4l2copy from [v4l2tools](https://github.com/mpromonet/v4l2tools) : 

          v4l2copy /dev/video0 /dev/video1 
This will configure /dev/video1 with the same format than /dev/video0 and copy all the frames.

Using gstreamer this could be done using:

         gst-launch v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1

Best Regards,
Michel.

nicogarcia, 2015-12-02:
Thanks for your quick response!

As you suggested I used v4l2copy but I got this from `v4l2-ctl --list-formats -d /dev/video1`:

```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'H264'
	Name        : [H264]
```

So I can't capture a JPEG image. I think I'm missing something: Is it possible to take JPEG and H264 from the same device? I think I should see JPEG also in that format list (or maybe all formats from raspi camera).

Thanks!

mpromonet, 2016-01-04:
Hi,  
I did not check the MMAL interface that is used by the V4L2 driver but I think it cannot be opened twice.
If you try to run twice `raspivid -t 0`, the second fails reporting : 

    mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
    mmal: camera component couldn't be enabled
    mmal: main: Failed to create camera component
    mmal: Failed to run camera app. Please check for firmware updates

Regards,
Michel.

closed 2016-01-04 by mpromonet

nicogarcia, 2016-01-04:
Ok, I'll try to get the picture from the stream.

Thank you very much for your response!

Happy new year!