mpromonet/v4l2rtspserver #15: Stuck at compressing /dev/video0 /dev/video10

crogerjack, 2016-02-23:
[image]

crogerjack, 2016-02-23:
[image]

mpromonet, 2016-02-23:
Hi,
This is more related to [v4l2tools](https://github.com/mpromonet/v4l2tools) project.
It could be nice to elaborate describing your problem, than posting picture.
Regards,
Michel.

crogerjack, 2016-02-23:
Hi, 
Thank you for your reply. 
Below are the output after I run it with -vv. 
I use vlc to open /dev/video10, it is successfully opened but keep buffering 0%

ubuntu@ubuntu-lenovo:~$ v4l2compress_h264 /dev/video0 /dev/video10 -vv
2016-02-23 16:20:09,901 [INFO  ] - /home/ubuntu/Downloads/v4l2wrapper/inc/logger.h:44
	level:DEBUG
2016-02-23 16:20:09,901 [NOTICE] - src/V4l2Capture.cpp:86
	driver:uvcvideo 85000001
2016-02-23 16:20:09,901 [NOTICE] - src/V4l2Capture.cpp:94
	/dev/video0 support read/write
2016-02-23 16:20:09,901 [NOTICE] - src/V4l2Capture.cpp:95
	/dev/video0 support streaming
2016-02-23 16:20:09,910 [NOTICE] - src/V4l2Capture.cpp:153
	/dev/video0:YUYV size:640x480 bufferSize:614400
2016-02-23 16:20:09,918 [NOTICE] - src/V4l2Capture.cpp:172
	fps:1/30
2016-02-23 16:20:09,918 [NOTICE] - src/V4l2Capture.cpp:173
	nbBuffer:0
2016-02-23 16:20:09,918 [NOTICE] - src/V4l2Output.cpp:41
	Output device name:v4l2 loopback cap:85008003
2016-02-23 16:20:09,918 [NOTICE] - src/V4l2MmapDevice.cpp:55
	Device /dev/video10 nb buffer:8
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:0 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:1 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:2 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:3 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:4 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:5 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:6 size:1228800
2016-02-23 16:20:09,918 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video10 buffer idx:7 size:1228800
2016-02-23 16:20:09,918 [NOTICE] - src/v4l2compress_h264.cpp:117
	Start Capturing from /dev/video0
2016-02-23 16:20:09,919 [NOTICE] - src/V4l2MmapDevice.cpp:55
	Device /dev/video0 nb buffer:10
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:0 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:1 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:2 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:3 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:4 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:5 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:6 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:7 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:8 size:614400
2016-02-23 16:20:09,919 [INFO  ] - src/V4l2MmapDevice.cpp:74
	Device /dev/video0 buffer idx:9 size:614400
x264 [info]: using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX AVX2 FMA3 LZCNT BMI2
x264 [info]: profile Constrained Baseline, level 2.2
2016-02-23 16:20:10,204 [NOTICE] - src/v4l2compress_h264.cpp:172
	Start Compressing /dev/video0 to /dev/video10
x264 [debug]: frame=   0 QP=20.00 NAL=3 Slice:I Poc:0   I:1200 P:0    SKIP:0    size=29557 bytes
2016-02-23 16:20:10,269 [INFO  ] - src/v4l2compress_h264.cpp:224
	Copied 0/4 size:27

mpromonet, 2016-02-23:
It seems it hangs on the second write.
The `-w` option that should allow to use write interface instead of mmap api was not working.
Since the commit 8b6183c786438e34686bd964dc294aad3879f88d the option is decoded.
Perhaps the version of v4l2loopback used doesnot manage correctly the H264 using the mmap API, I submit pull request to fix this.
You can try using '-w' option.
Regards
Michel.

crogerjack, 2016-02-24:
I try on another laptop and the v4l2compress_h264 /dev//video0 /dev/video10 working fine, but when I run the h264_v4l2_rtspserver /dev/video10 and then use vlc to play the rtsp stream, the vlc shows debug message: cannot decode one frame.
I also use gstreamer to play video from /dev/video10, but the gstreamer only able to display the video for one second and then stopped. Below are the gstreamer code that I use.
gst-launch-1.0 -vvv v4l2src device=/dev/video10 ! queue ! video/x-h264, width=640, height=480, framerate=30/1 ! h264parse ! avdec_h264 ! xvimagesink sync=false

mpromonet, 2016-05-20:
This thread should takes place in v4l2tools project.

closed 2016-05-20 by mpromonet