mpromonet/v4l2rtspserver #16: Unable to stream from sever using Matlabs camera IP toolbox

BrownKyle, 2016-02-24:
I have a raspberry pi set up to stream data using this rtsp server. I am able to stream the data from an OSX terminal, with great performance with:
`netcat -l -p 5000 | mplayer -fps 60 -cache 1024 rtsp://192.168.1.1:8554/unicast`

I would like to import the data the live video stream into Matlab. Matlab provides a toolbox to do such. However, Matlab will crash if I request to stream a video without a file extension. I have tried a few, .h264, .mp4 etc. But none seem to work. What is the appropriate file extension for the video stream?

mpromonet, 2016-02-24:
Hi,
The rtsp streamer send, by default, H264 Elementary stream, then the quite usual extension is .h264.
In order to change the stream url you can user `-u` option: 

     h264_v4l2_rtspserver -u video.h264
will create a streaming session that could be requested using : 

     rtsp://192.168.1.1:8554/video.h264

If you need transport stream you can use `-T` option and use .ts extension : 
     h264_v4l2_rtspserver -u video.ts -T

Best Regards,
Michel.

BrownKyle, 2016-02-24:
Thanks Michel.

I appreciate you answering that.

I trailed both the methods you suggested, Matlab would crash when I gave it the .h264 extension after adding the `-u` option. Previously it did not crash when this file extension was given, it only crashed when no file extension was given. I took the crash to be caused when Matlab successfully detects the incoming stream. I looked into the ipcam() function in Matlab and found it only accepts RTSP video when the video has Motion JPG codec.

I think I'll need to find another way of sending the raspberry pi video so that its a different codec.

Your software was very impressive low latency stream, if only Matlab was a little less picky when it came to the codecs of videos sent over RTSP.

Regards
Kyle

closed 2016-02-24 by BrownKyle