mpromonet/v4l2rtspserver #17: Can't view stream on first launch

MrJacK67, 2016-02-24:
Hi,

On a RPI2 with latest Raspbian Jessie, when i start the rtspserver the first time, vlc doesn't play anything, no error message.

> pi@raspicam:~ $ sudo /home/pi/h264_v4l2_rtspserver/h264_v4l2_rtspserver -P 8554 -Q 10 -r -s -W 1920 -H 1080 -F 25
> 2016-02-24 14:57:29,910 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:243
> 	Create V4L2 Source.../dev/video0
> 2016-02-24 14:57:30,265 [NOTICE] - src/V4l2Capture.cpp:86
> 	driver:raspicam 1000001
> 2016-02-24 14:57:30,265 [NOTICE] - src/V4l2Capture.cpp:94
> 	/dev/video0 support read/write
> 2016-02-24 14:57:30,493 [NOTICE] - src/V4l2Capture.cpp:153
> 	/dev/video0:H264 size:1920x1080 bufferSize:262144
> 2016-02-24 14:57:30,502 [NOTICE] - src/V4l2Capture.cpp:172
> 	fps:1/25
> 2016-02-24 14:57:30,503 [NOTICE] - src/V4l2Capture.cpp:173
> 	nbBuffer:4
> 2016-02-24 14:57:30,504 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:263
> 	Start V4L2 Capture.../dev/video0
> 2016-02-24 14:57:30,505 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:85
> 	Play this stream using the URL "rtsp://192.168.67.152:8554/unicast"
> RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

The last line appear when VLC connect.

I stop the program with CTRL + C, launch it again and now :

> pi@raspicam:~ $ sudo /home/pi/h264_v4l2_rtspserver/h264_v4l2_rtspserver -P 8554 -Q 10 -r -s -W 1920 -H 1080 -F 25
> 2016-02-24 15:03:10,829 [ERROR ] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:232
> 	Failed to create RTSP server: bind() error (port number: 8554): Address already in use
> 

So i wait a moment and i can start it again, now VLC start to play the stream immediately

MrJacK67, 2016-02-24:
I see a little difference between the first and second start :+1: 

> 2016-02-24 15:24:40,398 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:85
> 	Play this stream using the URL "rtsp://192.168.67.152:8554/unicast"
> RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

Second :

> 2016-02-24 15:27:55,042 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:85
> 	Play this stream using the URL "rtsp://192.168.67.152:8554/unicast"
** 2016-02-24 15:27:55,253 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:84
> 	profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQDwBE/LAPEiagA==,KO4CXLA=**
> RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

mpromonet, 2016-02-24:
Hi,
In order to be played an H264 stream needs SPS, PPS information.
Depending on the V4L2 device driver, it is needed to wait for it.
In order to investigate what frame are got from V4L2 device, you can use the `-v` argument.
Regards,
Michel.

MrJacK67, 2016-02-24:
I don't have this at first launch :

> 2016-02-24 23:26:43,305 [INFO  ] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/inc/logger.h:44
>         level:INFO
> 2016-02-24 23:26:43,314 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:243
>         Create V4L2 Source.../dev/video0
> 2016-02-24 23:26:43,634 [NOTICE] - src/V4l2Capture.cpp:86
>         driver:raspicam 1000001
> 2016-02-24 23:26:43,635 [NOTICE] - src/V4l2Capture.cpp:94
>         /dev/video0 support read/write
> 2016-02-24 23:26:43,900 [NOTICE] - src/V4l2Capture.cpp:153
>         /dev/video0:H264 size:1920x1080 bufferSize:262144
> 2016-02-24 23:26:43,901 [NOTICE] - src/V4l2Capture.cpp:172
>         fps:1/25
> 2016-02-24 23:26:43,901 [NOTICE] - src/V4l2Capture.cpp:173
>         nbBuffer:4
> 2016-02-24 23:26:43,901 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:263
>         Start V4L2 Capture.../dev/video0
> 2016-02-24 23:26:43,902 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:85
>         Play this stream using the URL "rtsp://192.168.0.18:8554/unicast"
> 2016-02-24 23:26:44,407 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456352804 fps:1 bandwidth:110kbps
> 2016-02-24 23:26:45,032 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456352805 fps:17 bandwidth:7068kbps
> 2016-02-24 23:26:46,032 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456352806 fps:25 bandwidth:12011kbps
> 2016-02-24 23:26:46,636 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
>         IDR size:94858
> 2016-02-24 23:26:47,031 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456352807 fps:25 bandwidth:12249kbps
> 2016-02-24 23:26:48,030 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456352808 fps:25 bandwidth:11943kbps

Second ok :

> 2016-02-24 23:30:26,440 [INFO  ] - /home/pi/h264_v4l2_rtspserver/v4l2wrapper/inc/logger.h:44
>         level:INFO
> 2016-02-24 23:30:26,443 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:243
>         Create V4L2 Source.../dev/video0
> 2016-02-24 23:30:26,640 [NOTICE] - src/V4l2Capture.cpp:86
>         driver:raspicam 1000001
> 2016-02-24 23:30:26,640 [NOTICE] - src/V4l2Capture.cpp:94
>         /dev/video0 support read/write
> 2016-02-24 23:30:26,912 [NOTICE] - src/V4l2Capture.cpp:153
>         /dev/video0:H264 size:1920x1080 bufferSize:262144
> 2016-02-24 23:30:26,914 [NOTICE] - src/V4l2Capture.cpp:172
>         fps:1/25
> 2016-02-24 23:30:26,915 [NOTICE] - src/V4l2Capture.cpp:173
>         nbBuffer:4
> 2016-02-24 23:30:26,917 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:263
>         Start V4L2 Capture.../dev/video0
> 2016-02-24 23:30:26,918 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/main.cpp:85
>         Play this stream using the URL "rtsp://192.168.0.18:8554/unicast"
> 2016-02-24 23:30:27,113 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456353027 fps:1 bandwidth:0kbps
> 2016-02-24 23:30:27,114 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:57
>         SPS size:16
> 2016-02-24 23:30:27,114 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:58
>         PPS size:5
> 2016-02-24 23:30:27,115 [NOTICE] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:84
>         profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQDwBE/LAPEiagA==,KO4CXLA=
> 2016-02-24 23:30:27,235 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
>         IDR size:9593
> 2016-02-24 23:30:28,003 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456353028 fps:20 bandwidth:7031kbps
> 2016-02-24 23:30:29,002 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456353029 fps:25 bandwidth:12013kbps
> 2016-02-24 23:30:29,647 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/H264_V4l2DeviceSource.cpp:59
>         IDR size:92404
> 2016-02-24 23:30:30,002 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456353030 fps:25 bandwidth:12082kbps
> 2016-02-24 23:30:31,003 [INFO  ] - /home/pi/h264_v4l2_rtspserver/src/V4l2DeviceSource.cpp:29
>         intv_sec:1456353031 fps:25 bandwidth:11832kbps

mpromonet, 2016-02-26:
Hi,
From the log, the first time the V4L2 driver seems to not send the SPS and PPS frame that are needed to play the stream.
This depend on the driver you are using.
With the official raspicam driver `bcm2835_v4l2`, it send these 2 frames at the beginning and repeat it at a period controlled by `h264_i_frame_period` parameter, you can check the value with:

     v4l2-ctl -C h264_i_frame_period

Regards,

Michel.

MrJacK67, 2016-03-10:
Thank you Michel, the driver was the problem !

closed 2016-03-10 by MrJacK67

lukasz-zaroda, 2016-09-09:
Ok, so... how to fix this problem with the driver? I'm in the same situation :) .

mpromonet, 2016-09-09:
Setting h264_i_frame_period

alex88, 2017-04-24:
@mpromonet I'm in the same situation, is there a reason why I never get the SPS or PPS on the first try but I have to start for another one? Is there a timeout after which the server doesn't try to read it anymore? On first start even after a minute it doesn't find it, on second, almost immediately

mpromonet, 2017-04-29:
Hi,
Did you tried to change the h264_i_frame_period ? 
It is possible that the parsing to get SPS/PPS fails, did you look a the verbose log to see what's occurs ?
Regards,
Michel.

alex88, 2017-04-29:
Unfortunately the webcam (Microsoft HD 3000) doesn't have that control, however I've solved for now using ffmpeg+ffserver on an rPI and ffmpeg on the other one