mpromonet/v4l2rtspserver #18: Port reuse

ronnyvdbr, 2016-03-01:
Hi, i'm still using your program and am implementing it currently in my web cam project.
http://ronnyvdbr.github.io/RaspberryIPCamera/

When I launch your executable and close it, then I'm not able to immediately re-launch it.
I get the following message:
Failed to create RTSP server: bind() error (port number: 8554): Address already in use

I should be able to overcome this if I compile your work with the following option:
ALLOW_RTSP_SERVER_PORT_REUSE=1

I have used this option before when compiling live555 rtsp server as per below procedure (note the option ALLOW_RTSP_SERVER_PORT_REUSE=1)

cd /home/pi/
wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz
tar -zxvf live555-latest.tar.gz
mv live live555
cd live555
sed -i 's/-D_FILE_OFFSET_BITS=64/-D_FILE_OFFSET_BITS=64 -DALLOW_RTSP_SERVER_PORT_REUSE=1/g' ./config.linux
./genMakefiles linux
make

Would u have any idea where I need to insert this option to also use it to compile your work?

ronnyvdbr, 2016-03-01:
Never mind, found it:

cmake . && make CFLAGS='-DALLOW_RTSP_SERVER_PORT_REUSE=1'

ronnyvdbr, 2016-03-02:
Hmm, apparently the compile options didn't resolve the issue.
Below is a link to a mailing list with a solution description.
http://lists.live555.com/pipermail/live-devel/2011-October/013966.html

Do u think this piece of code is present in your work?

mpromonet, 2016-03-03:
Hi Ronny,
The option should be set compiling live555, not compiling h264_v4l2_rtspserver.
However perhaps it is possible to close properly the RTSPServer instance, I will make some tries but I am afraid that the socket is closed without reading the pending data.
Best Regards,
Michel.

ronnyvdbr, 2016-03-04:
Hi, thanks for your reply.
I tried to narrow down the usage of the SO_REUSEADDR and SO_REUSEPORT
socket options in h264_v4l2_rtspserver but wasn't able to find these socket
options in your source code.
I was however able to find traces of this in groupsock.hh from the live555
RTSP server.
Perhaps your design uses another version of the groupsock.hh header file.

I have created a systemd service unit file to start your software, but each
time I restart it in my php code, the error 'the address or port is already
in use' will rise.  An end user does not see that info and only concludes
it's just not working unfortunately.
As a test I tried to stop your software via sighup instead of sigkill but
the outcome stays the same.

Hope it helps u narrowing down the issue as I can't find any other RTSP
server without other issues.

mpromonet, 2016-03-05:
Hi Ronny,
In order to build live555 with the `ALLOW_RTSP_SERVER_PORT_REUSE` flag, you could use : 

    wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz
    tar -xzf live555-latest.tar.gz
    cd live
    ./genMakefiles linux
    sudo make CPPFLAGS=-DALLOW_RTSP_SERVER_PORT_REUSE=1 install
This will install live555 in /usr/local.

Next you could build `h264_v4l2_rtspserver` with it : 

    git clone https://github.com/mpromonet/h264_v4l2_rtspserver
    cd h264_v4l2_rtspserver
    cmake CMAKE_PREFIX_PATH=/usr/local . && make

Regards,
Michel.

ronnyvdbr, 2016-03-05:
Hi Michel, I'm going to try this in the next couple of hours, it will make
the system so much better.
Thanks for taking the time to search for a solution for this issue, really
appreciate it.

mpromonet, 2016-05-17:
Hi Ronny,
Did you succeeded to set the reuse port option ?
If it works properly, I will add a note in the readme and close this issue.
Regards,
Michel.

ronnyvdbr, 2016-05-17:
Hi, yes I was able to solve this as per your compiling instructions.
I have built a web interface for it using your rtsp server, php and
bootstrap.
You can find the design at https://ronnyvdbr.github.io/RaspberryIPCamera/
Thanks for your help with this.

mpromonet, 2016-05-20:
Hi Ronny,
Nice interface.
Regards,
Michel.

closed 2016-05-20 by mpromonet

shelteroperations, 2019-03-28:
> Failed to create RTSP server: bind() error (port number: 8554): Address already in use

I don't use the 555 software, just v4l2rtspserver on Raspberry Pi but I wanted to note something that fixed similar problem for me. I have a script that stops (systemctl stop v4l2rtspserver) so it can take a picture, then starts again. Every time almost, it says "port still in use." 8554 I think it's using.

Anyway....just put this Before the "systemctl start" and at least for me it worked. Almost like it "clears out" the ports that were told to close but still were open? Anyway it works....

netstat -tulp

(I didn't install that 555 software I think so maybe you need that but I am not starting it or using it on purpose, just v4l2rtspserver.