mpromonet/v4l2rtspserver #22: Cannot start HTTP server

MichaelAronsen, 2016-05-24:
When starting with -T option to start the HTTP/RTSP server, server is still started on the RTSP only port (default 8554):
root@indoorpi:/home/pi# v4l2rtspserver -T 8000 -f /dev/video0
2016-05-24 14:18:52,153 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:297
	Create V4L2 Source...8000
VIDIOC_REQBUFS: Inappropriate ioctl for device
2016-05-24 14:18:52,165 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:316
	Create Source ...8000
2016-05-24 14:18:52,176 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:111
	Play this stream using the URL "rtsp://192.168.1.244:8554/8000/unicast"
2016-05-24 14:18:52,177 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:297
	Create V4L2 Source.../dev/video0
2016-05-24 14:18:52,180 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:94
	begin thread
2016-05-24 14:18:52,181 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:207
	V4L2DeviceSource::getNextFrame no data errno:0 Success
2016-05-24 14:18:52,181 [ERROR ] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
	error:Success
2016-05-24 14:18:52,182 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
	end thread
2016-05-24 14:18:52,185 [NOTICE] - src/V4l2Device.cpp:131
	driver:bm2835 mmal 85200005
2016-05-24 14:18:52,191 [NOTICE] - src/V4l2Device.cpp:139
	/dev/video0 support read/write
2016-05-24 14:18:52,196 [NOTICE] - src/V4l2Device.cpp:141
	/dev/video0 support streaming
2016-05-24 14:18:52,202 [NOTICE] - src/V4l2Device.cpp:199
	/dev/video0:H264 size:1296x972 bufferSize:1280512
2016-05-24 14:18:52,208 [NOTICE] - src/V4l2Device.cpp:220
	fps:1/25
2016-05-24 14:18:52,210 [NOTICE] - src/V4l2Device.cpp:221
	nbBuffer:1
2016-05-24 14:18:52,231 [NOTICE] - src/V4l2MmapDevice.cpp:54
	Device /dev/video0 nb buffer:10
2016-05-24 14:18:52,703 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:316
	Create Source .../dev/video0
2016-05-24 14:18:52,708 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:111
	Play this stream using the URL "rtsp://192.168.1.244:8554/video0/unicast"
2016-05-24 14:18:52,712 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:94
	begin thread
2016-05-24 14:18:52,719 [NOTICE] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
	profile-level-id=     1;sprop-parameter-sets=AAAAASdkACCsK0Aog9/cA8SJqA==,AAAAASjuAlyw
^CSIGINT
2016-05-24 14:18:55,005 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:371
	Exiting....
2016-05-24 14:18:55,015 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2016-05-24 14:18:55,015 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2016-05-24 14:18:55,016 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2016-05-24 14:18:55,016 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames

mpromonet, 2016-05-24:
Hi Michael,
The -T option add a Transport Stream muxer, it doesnot add an HTTP server.
I realize that options described in README are out of date, you should look to '-h' usage.
In order to enable RTP/RTSP/HTTP you should use '-p 8000'.
Best Regards,
Michel.

MichaelAronsen, 2016-05-25:
That is actually what I did:

pi@roofpi:~ $ v4l2rtspserver -h
v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]
	          [-I interface] [-P RTSP port] [-T RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout]
	          [-r] [-w] [-s] [-W width] [-H height] [-F fps] [device] [device]

But thanks for the reply, will play around with it :)

closed 2016-05-25 by mpromonet

pepeRossRobotics, 2020-06-05:
Hello,
Is there any example on how to launch the http server?
I am currently using this command:
`./v4l2rtspserver -I 10.42.0.155 -p 8080 -S2 -P 8081 -H 768 -Q 1 -W 1024 -F 30 /dev/video0`

```
log level:500

[NOTICE] /home/pi/v4l2rtspserver/main.cpp:385
	Version: 0.2.0-17-g9810118 live555 version:2020.05.15
[NOTICE] /home/pi/v4l2rtspserver/main.cpp:426
	Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:uvcvideo capabilities:84200001 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video0 support streaming
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:202
	Cannot set pixelformat to:H264 format is:YUYV
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:215
	/dev/video0:MJPG size:1024x768 bufferSize:1573453
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:236
	fps:1/30
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:237
	nbBuffer:0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
[NOTICE] /home/pi/v4l2rtspserver/main.cpp:449
	Create Source .../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:82
	Play this stream using the URL "rtsp://10.42.0.155:8081/ts"
[NOTICE] /home/pi/v4l2rtspserver/main.cpp:527
	HLS       http://10.42.0.155:8081/ts.m3u8
[NOTICE] /home/pi/v4l2rtspserver/main.cpp:528
	MPEG-DASH http://10.42.0.155:8081/ts.mpd
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:82
Play this stream using the URL "rtsp://10.42.0.155:8081/unicast"SIGINT
```

I get a really nice RSTP that I can see with VLC using
`vlc --network-caching=10 rtsp://10.42.0.155:8081/video0/unicast`
But when I try to see the http, it does not show neither on the suggested webpage  ot on vlc.
Is there any exmaple on how to enable the Http stream and see it?

commit 8ab95d26 (2016-05-25): fix #22 usage inconsistency for HTTP port option