mpromonet/v4l2rtspserver #23: MJPEG support

MichaelAronsen, 2016-05-25:
Hi,

Not sure this should be an issue, but rather a question.

But since v4l2 device in the pi now has mjpeg support, would it be possible to simply stream the raw mjpeg stream from v4l2 over rtsp using v4l2rtspserver?

Reason I ask is that mjpeg would allow for higher resolution and less compression.

Regards,
Michael

mpromonet, 2016-05-25:
Hi,

This works, but the bandwidth is much higher.
I don't undersand why do you think jpeg capture allow higher resolution, the compression is done after capture.

Regards,
Michel.

MichaelAronsen, 2016-05-25:
It allows higher resolution as there is some limit in the pi GPU code that
limits h264 to max 1920x1080, where as mjpeg is not limited.

See this post:
http://www.lewisroberts.com/2015/05/15/raspberry-pi-mjpeg-at-30fps/

MichaelAronsen, 2016-05-26:
How would I enable mjpeg streaming rather than h264?

mpromonet, 2016-05-26:
using -fJPEG

prechtl, 2016-06-18:
I am trying to implement this mjpeg encoding as well.  I have taken the basic instructions and made the following changes:

I set the original uv4l command using:
`sudo uv4l --driver raspicam --auto-video_nr --encoding mjpeg`

Then as you suggest, I call the server w/ the -fJPEG option using
`./v4l2rtspserver -s -fJPEG`

However, when I include the '-fJPEG' option and open my VLC terminal to view the rtsp://<url>:8554/unicast stream, my VLC window just jumps back and forth between two sizes - but no video shows up.  Whereas if I eliminate the '-fJPEG' option I get video to show up.  Problem is that when I view the 'Codec information' in VLC for this video, it ID's the Codec as: 'H264 - MPEG-4 AVC (part 10) (h264)'  So it's unclear to me that this is using mjpeg encoding ... seems tied to H264.

(fyi: I'm interested in avoiding H264 b/c of the parallel question I have asked in this github project where I'm trying to capture full field-of-view video that uses all 2592 x 1944 pixels of the raspberry pi camera)

mpromonet, 2016-06-19:
Hi prechtl,

If you are using uv4l (that is closed source), I guess you should use `--encoding jpeg`. Generaly I prefer to use the official V4L2 driver `bcm2835-v4l2`

Using `-s` option is probably a bad idea, it will read from the V4L2 capture device and send to network in the same thread.

Regards,
Michel.

closed 2016-07-15 by MichaelAronsen