mpromonet/v4l2rtspserver #27: Stream to file

prechtl, 2016-06-18:
Thanks so much for this!  Is there a way to adapt this to stream to a file at 2592x1944 resolution (at the max possible frame rate) on the Raspberry Pi?

mpromonet, 2016-06-19:
What do you mean by Stream to file ?    
Are you asking from reading a file and streaming throught RTSP ?

prechtl, 2016-06-23:
Basically, I want to send the video stream to a file - ideally storing the full-field-of-view video at 10 or 15 frames-per-second to a file.  I don't think the raspberry Pi can output 2592x1944 video - but I do believe it can send out a downsampled version of that full field to output 1920x1440 video at these frame rates.  If we assume 12 bits per pixel and 10 FPS and a nominal compression ratio of 30, the data rate will be about 11 Mbps.  The 1st question is whether the SD Card can be written to at that rate.  The 2nd question is what software to use to do the writing.  I will look into openRTSP

mpromonet, 2016-06-27:
If I understand well, you don't need to stream the video, using `raspistill` will save JPG video stream to a file.

prechtl, 2016-06-28:
I found some discussion online at https://www.raspberrypi.org/forums/viewtopic.php?t=45178 that talks about this.  I am not able to test it right now.  Will be away from my RPi hardware for a short period.   I will try this when I'm in front of the hardware again.  My concern/suspicion is that w/ raspistill, the frame rate I'll get will be slow (~ 1 FPS).  It's just that I'm currently able to 'stream' at 10 FPS over the network.  All I really need is a way to record that stream of data.  I'm trying to figure out the easiest way to do that.  I don't want to record for hours - just 10 seconds up to a minute or two at a time...

mpromonet, 2016-06-28:
raspistill use `mmal` interface that is also used by v4l2 driver (and probably by closed-source `uv4l`). 
Then you will probably not get better result than with `raspistill`

prechtl, 2016-07-14:
I think I am able to do about what I want by using v4l2 and cvlc.  Specifically, I can issue a command like
`cvlc    --no-audio v4l2:///dev/video0 --v4l2-width 1920 --v4l2-height 1440 \
        --v4l2-chroma MJPG --run-time=10 --sout file/ogg:/tmp/stream_v01.mpg \
        vlc://quit` 

and it records a stream of video for 10 seconds at 1920x1440.  But, b/c of the settings used for v4l2, these pixels will be distributed above the full FOV of the image sensor.  And I can get frame rates approaching 10 FPS.

closed 2016-07-14 by prechtl