mpromonet/v4l2rtspserver #28: Could NOT find ALSA in CMake

lavajgh, 2016-06-29:
Trying to install on Pi3, totally new to this, and following CMAKE directions.   When I go to CMAKE the file, I am getting an error that i can not find the ALSA files.  If i do:

whereis alsa I get
alsa: /usr/share/alsa

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ALSA_LIBRARY (ADVANCED)
    linked by target "h264_v4l2_rtspserver" in directory /home/pi/live/h264_v4l2_rtspserver

mpromonet, 2016-06-29:
Hi lavajgh,
I recently start some ALSA support (that are ongoing), in order to build, you need to install libasound2-dev using : 

       sudo apt-get install libasound2-dev
I will upgrade the README about this new dependency.

Best Regards,
Michel.

lavajgh, 2016-06-29:
Thank you Michel.  I followed the YML script that you had lined by line which has the dependancy already and it worked.  Thank you for the work on this.

closed 2016-06-29 by lavajgh