mpromonet/v4l2rtspserver #29: Request: authentication support

roger-, 2016-06-29:
Would be useful if RTSP streams could be password protected.

mpromonet, 2016-06-29:
Hi Roger,
This is easy to add, live555 include this feature.
It just need few lines of code before creating the RTSPServer.
Regards,
Michel.

closed 2016-07-10 by mpromonet

LeonardoGentile, 2017-02-07:
Thanks for this 👍 
But how to use it? Could you please add few lines to the Readme explaining how to set the password on the server side and pass it from the client side?

mpromonet, 2017-02-07:
Hi Leonardo,
You're right some details could helps... 
	std::cout << "\t -U user:password : RTSP user and password" 
	std::cout << "\t -R realm : use md5 password 'md5(<username>:<realm>:<password>')" << std::endl;

Shortly you can define plain text password using : 
       -U user1:password1 -U user2:password2
or you can use md5 password using : 
       -R realm -U user1:575b76a870180382669998eb63a60426 
For the client side, it depends on the client some accept url like : 
      rtsp://user1:password1@.....

Regards,
Michel.

LeonardoGentile, 2017-02-08:
thanks, it works! 👍

cedricdev, 2017-05-09:
Hi.
Sorry to ask a question a year later...
The  plain text authentification works very well but, i can not use md5 authentification.
i tried that :
`-R realm -U user1:"md5(<username>:<realm>:<password>)"`
and 
`-R realm -U "md5(<username>:<realm>:<password>)"`

of course when i write : 
`md5(<username>:<realm>:<password>)` it's the result of

"user:raspberrypi:pass" on https://www.md5.fr/
raspberrypi is the hostname of my IP cam.

I d'ont know if there is an issue or if i'm wrong

mpromonet, 2017-05-10:
Hi,
The realm should be the one you used.
If you use raspberrypi for realm, the command line option should be : 

     - R rasberrypi -U "md5(<username>:raspberrypi:<password>)"

Best Regards,
Michel.

cedricdev, 2017-05-14:
Thanks for your reply.
I can not use it. maybe i don't understand what is realm. (hostname of my raspberry?)

here my commandline : 
`sudo /home/pi/git/v4l2rtspserver/v4l2rtspserver -F 15 -R raspberrypi -U e73c7c9f7959d76630c2385436b364a1 -W 1280 -H 720 -P 8555 /dev/video0`

e73c7c9f7959d76630c2385436b364a1 is the result of `md5(user:raspberrypi:pass)`

when i try to connect at : 
`rtsp://192.168.0.90:8555/unicast`
it asks me user and password. it doesn't work.

That is not a big problem because i use the plain text for credantial.

mpromonet, 2017-05-14:
Hi,
You were very closed to the solution, you should use : 

    -R raspberrypi -U user:e73c7c9f7959d76630c2385436b364a1

Best Regards,
Michel.

cedricdev, 2017-05-14:
It works! Merci pour ces précisions!
Et encore bravo pour ce super boulot. votre solution est la meilleur et la plus stable pour faire tourner une camera ip raspberry sur mon synology.

mpromonet, 2017-05-14:
Hi,
This was the reason to create this project as I did not find any suitable solution to stream the rasperry camera. All the tutorial using a pipe between raspivid and vlc was not a nice solution neither for latency, nor for CPU :)
Best Regards,
Michel.

lubinopetrillo, 2018-05-26:
Hi,

Just to clarify things a little bit...

If you want to use a **plain text password**, you can change how the server starts (_start()_ function) in the **rtsp-h264 file** (under the _controlscripts_ folder), like that: 

`/system/sdcard/bin/busybox nohup /system/sdcard/bin/v4l2rtspserver-master $RTSPH264OPTS -U foo:bar -U user:plaintextpassword  >> "$LOGPATH" &`

Now, if you want to use **md5 authentication**, you should change the same line adding the corresponding md5 parameters:

`/system/sdcard/bin/busybox nohup /system/sdcard/bin/v4l2rtspserver-master $RTSPH264OPTS -R dafanghacks -U user:md5sum  >> "$LOGPATH" &`

In order for it to work, you have to use in this case _dafanghacks_ as the realm (as it is written in the line above), and the md5sum is the result of calculating the MD5 of _username:realm:password_.

After that you can try to connect to the RTSP stream with any app and log in using your user name and password. 

Best regards.

MeansonChen, 2023-04-19:
Hi,
Excuse me, I have a question about authentication.
what is the different between use string pass word and MD5 password.
I capture some network message by wireshark, and i can not find different between them.

MeansonChen, 2023-04-19:
Hi,
Sorry to bother you, I have a question about authentication.
what is the different between use string pass word and MD5 password.
I capture some network message by wireshark, and i can not find different between them.

mpromonet, 2023-04-19:
Hi @MeansonChen 

The difference should be with md5 password, the real password is not exchanged.

Best Regards
Michel

commit f8de2847 (2016-07-10): fix #29: add authentification support