mpromonet/v4l2rtspserver #31: Always resetting frame rate to 25 on restart

patmu, 2016-08-13:
Hello

According to manual, if I start v4l2rtspserver with parameter '-f', it should use the current v4l2 driver settings for width, heigth and frame rate.

This works fine for width and hight, but not for frame rate. After (re-)starting v4l2rtspserver the frame rate is always reset to 25.

Could this be a bug?

Regards,
Pat

mpromonet, 2016-08-13:
Hi Pat,
You are right, using '-f' ignore VIDIOC_S_FMT ioctl so '-W' '-H', but it doesnot ignore VIDIOC_S_PARM ioctl.
In order to disable fps setting, you should specify "-F 0".
Regards,
Michel.

closed 2016-08-13 by mpromonet

patmu, 2016-08-13:
Hi Michel

That does the trick!

Specify "-F 0", the frame rate isn't resetting anymore.

Thanks for your (very) fast answer!

Regards,
Pat

commit af18d537 (2016-08-13): fix #31: update command line description indicating -F0 disable fps setting