mpromonet/v4l2rtspserver #32: SDP seems too big

idreamerhx, 2016-08-26:
2016-08-26 11:04:20,664 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
	profile-level-id= 20603;sprop-parameter-sets=BwIGAwQFAQ.........................AAADUAAAD/////AgAAAAAAAAAAAAAAAA==

This always a huge output, and ...

when I use openRTSP:
openRTSP rtsp://192.168.1.8:8554/unicast

OUTPUT:
Sending request: DESCRIBE rtsp://192.168.1.8:8554/unicast RTSP/1.0
CSeq: 3
User-Agent: openRTSP (LIVE555 Streaming Media v2016.08.07)
Accept: application/sdp

Received 19999 new bytes of response data.
Failed to get a SDP description for the URL "rtsp://192.168.1.8:8554/unicast": Response buffer size (20000) is too small for "Content-Length:" 60055 (need a buffer size of >= 60222 bytes

And ffmpeg seems blocked by SDP.  This may be a LiveMedia issue.

idreamerhx, 2016-08-26:
Sorry, I forget to pull the latest code, let me try..

idreamerhx, 2016-08-26:
Received 19999 new bytes of response data.
Failed to get a SDP description for the URL "rtsp://192.168.1.8:8554/unicast": Response buffer size (20000) is too small for "Content-Length:" 579399 (need a buffer size of >= 579567 bytes

mpromonet, 2016-08-27:
Hi idreamerhx,
This is probably because the code that try to extract SPS,PPS (that are usually got in with a picture) failed to split it with marker.
Adding the argument '-v' you should see the SPS,PPS size.
Regards,
Michel.

closed 2016-09-03 by idreamerhx