mpromonet/v4l2rtspserver #34: compiling fails at cmake

regisin, 2016-09-01:
Error when running "cmake . && make"

I have a Raspberry Pi 2 model B, running Linix 4.4.11 #888

CMakeError.log content:

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec255891337/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec255891337.dir/build.make CMakeFiles/cmTryCompileExec255891337.dir/build
make[1]: Entering directory '/var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec255891337.dir/CheckSymbolExists.c.o
/usr/bin/cc   -Wall    -o CMakeFiles/cmTryCompileExec255891337.dir/CheckSymbolExists.c.o   -c /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec255891337
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec255891337.dir/link.txt --verbose=1
/usr/bin/cc  -Wall     CMakeFiles/cmTryCompileExec255891337.dir/CheckSymbolExists.c.o  -o cmTryCompileExec255891337 -rdynamic
CMakeFiles/cmTryCompileExec255891337.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x38): undefined reference to CMakeFiles/cmTryCompileExec255891337.dir/build.make:88: recipe for target 'cmTryCompileExec255891337' failed
make[1]: Leaving directory '/var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp'
Makefile:118: recipe for target 'cmTryCompileExec255891337/fast' failed
`pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [cmTryCompileExec255891337] Error 1
make: *** [cmTryCompileExec255891337/fast] Error 2

File /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec1498311751/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1498311751.dir/build.make CMakeFiles/cmTryCompileExec1498311751.dir/build
make[1]: Entering directory '/var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1498311751.dir/CheckFunctionExists.c.o
/usr/bin/cc   -Wall -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec1498311751.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec1498311751
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1498311751.dir/link.txt --verbose=1
/usr/bin/cc  -Wall -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec1498311751.dir/CheckFunctionExists.c.o  -o cmTryCompileExec1498311751 -rdynamic -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec1498311751.dir/build.make:88: recipe for target 'cmTryCompileExec1498311751' failed
make[1]: Leaving directory '/var/tmp/v4l2rtspserver-0.0.1/CMakeFiles/CMakeTmp'
Makefile:118: recipe for target 'cmTryCompileExec1498311751/fast' failed
make[1]: *** [cmTryCompileExec1498311751] Error 1
make: *** [cmTryCompileExec1498311751/fast] Error 2
```

mpromonet, 2016-09-01:
Hi,
This is strange that cmake thread detection failed.
I have a raspberry 2 runnning Linux 4.4.19-v7 #906, and it build without problem.
From the error log it seems `libpthread.so` is missing on your system, quite strange ... this library is provided by libc6 that is probably installed in your system.
Are you able to build the simple code that is in the error log of cmake ?
Regards,
Michel.

regisin, 2016-09-03:
I don't know if it'll work on a Pi2, but now I have a Pi 3 model B. Was having the same problem, googled and found this: http://stackoverflow.com/questions/24813827/cmake-failing-to-detect-pthreads-due-to-warnings/25130590#25130590

After that, ASLA was giving errors so I removed the few lines in cmakelists. Then I ran cmake and make separately.

The problem now is that when I try to initialize, it complains (this is my terminal output):
'''
pi@raspberrypi:~/Downloads/v4l2rtspserver $ sudo ./v4l2rtspserver /dev/video0
2016-09-03 00:55:03,399 [NOTICE] - /home/pi/Downloads/v4l2rtspserver/src/main.cpp:348
	Create V4L2 Source.../dev/video0
2016-09-03 00:55:03,465 [NOTICE] - src/V4l2Device.cpp:131
	driver:uvcvideo 84200001
2016-09-03 00:55:03,465 [NOTICE] - src/V4l2Device.cpp:135
	/dev/video0 support streaming
2016-09-03 00:55:03,491 [ERROR ] - src/V4l2Device.cpp:180
	Cannot set pixelformat to:H264 format is:YUYV
2016-09-03 00:55:03,491 [ERROR ] - src/V4l2Device.cpp:81
	Cannot init device:/dev/video0
pi@raspberrypi:~/Downloads/v4l2rtspserver $
'''
I know for a fact the camera is working (I can stream using motion, but it sucks, that's why I want this to work!)

regisin, 2016-09-03:
UPDATE

I believe it is now capturing (the camera led is on, just like motion) (had to use the -f flag to work):

`sudo ./v4l2rtspserver -f /dev/video0`

Now I cannot connect to the rtsp://ip.add.re.ss:port/unicast. VLC says:
```
[00007fdf1c0011b8] live555 demux error: Failed to connect with rtsp://134.197.42.93:8888/unicast
[00007fdf140009b8] core input error: open of `rtsp://134.197.42.93:8888/unicast' failed
```

mpromonet, 2016-09-03:
Hi, 
This RTSP server need a V4L2 capture device in compress format (H264, VP8 or JPEG).
If you're device give YUYV (it seems a webcam with uvcvideo driver), you could  compress the video to an intermediate V4L2 device using [v4l2loopback](https://github.com/umlaeute/v4l2loopback).
This could be done [v4l2tools](https://github.com/mpromonet/v4l2tools) : 
 - v4l2compress_h264 compress from a V4L2 capture device in YUYV format to H264 using x264
 - v4l2compress_omx compresse from a V4L2 capture device in  YUV420 format to H264 using raspberry GPU  
Regards,
Michel.

regisin, 2016-09-04:
Installing this is becoming harder than I thought!

Compiling v4l2loopback turned out to be a real pain. In fact I don't think I was able to do that. Documentation also does not help. Can you provide more details how you made it work?

I also think this issue is done, no? Since I deviated from the topic so much...

mpromonet, 2016-09-04:
Compiling v4l2loopback need kernel header, but some distribution like debian or Ubuntu contains it. 
Then if you used raspbian you can try :
 
     sudo apt-get install v4l2loopback-dkms

closed 2016-09-18 by mpromonet