mpromonet/v4l2rtspserver #35: segment duration confused the http url

idreamerhx, 2016-09-03:
Hi, 

When I want to use m3u8, I first tried:
./v4l2rtspserver -S -W 640 -H 480 -F 20 /dev/video10
This use default (-S) segment duration value. And this works for me. 

But When I set some other value:
./v4l2rtspserver -S 10 -W 640 -H 480 -F 20 /dev/video10

The url and output make me confused and it can not be accessed by vlc or curl.
The oputput is like:

2016-09-03 23:00:02,607 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:348
	Create V4L2 Source...10
VIDIOC_REQBUFS: Inappropriate ioctl for device
2016-09-03 23:00:02,608 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:367
	Create Source ...10
2016-09-03 23:00:02,609 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:136
	Play this stream using the URL "rtsp://192.168.1.8:8554/10/unicast"
2016-09-03 23:00:02,609 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:410
	HLS       http://192.168.1.8:8554/10/unicast.m3u8
2016-09-03 23:00:02,609 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
	begin thread
2016-09-03 23:00:02,609 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:411
	MPEG-DASH http://192.168.1.8:8554/10/unicast.mpd
2016-09-03 23:00:02,609 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/V4l2DeviceSource.cpp:206
	V4L2DeviceSource::getNextFrame no data errno:0 Success

......
	
2016-09-03 23:00:02,617 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
	begin thread
2016-09-03 23:00:02,617 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:136
	Play this stream using the URL "rtsp://192.168.1.8:8554/video10/unicast"
2016-09-03 23:00:02,617 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:410
	HLS       http://192.168.1.8:8554/video10/unicast.m3u8
2016-09-03 23:00:02,617 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/main.cpp:411
	MPEG-DASH http://192.168.1.8:8554/video10/unicast.mpd
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 840451 bytes!
2016-09-03 23:00:06,266 [NOTICE] - /opt/mpromonet/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
	profile-level-id=     1;sprop-parameter-sets=AAAAASdkAB6sK0BQHtAPEiag,AAAAASjuHyw=

Both the two url can't play.

mpromonet, 2016-09-03:
Hi,
This is behavior of `getopt` when a option like `-S`could be used without argument and with argument, it doesnot support space between the option and the value.
You should use '-S10' and not '-S 10'
Best Regards,
Michel.

idreamerhx, 2016-09-05:
Thanks,

I see the getopt parameter is '......S::.....'. and I use -S10 works.

closed 2016-09-05 by idreamerhx