mpromonet/v4l2rtspserver #36: Error on compiling

ronnyvdbr, 2016-09-28:
Hi Michael, I was updating my camera image with the latest update of your RTSP server but i'm getting below error on compiling.   Do u think there is anything I can do to make this work?

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ALSA_LIBRARY (ADVANCED)
    linked by target "v4l2rtspserver" in directory /home/pi/v4l2rtspserver
LOG4CPP_INCLUDE_DIR
   used as include directory in directory /home/pi/v4l2rtspserver
LOG4CPP_LIBRARY
    linked by target "v4l2rtspserver" in directory /home/pi/v4l2rtspserver
V4L2WRAPPER_LIBRARY
    linked by target "v4l2rtspserver" in directory /home/pi/v4l2rtspserver

-- Configuring incomplete, errors occurred!
See also "/home/pi/v4l2rtspserver/CMakeFiles/CMakeOutput.log".
See also "/home/pi/v4l2rtspserver/CMakeFiles/CMakeError.log".

mpromonet, 2016-09-28:
Hi Ronny,

Did you build from a fresh clone repository ? If not you can remove cmake stuff using : 

    rm -rf CMakeCache.txt *.a CMakeFiles

The CMakeFile should try to install what is missing using `apt-get`.
If it doesnot work, you can try to install dependencies manually using:

    sudo apt-get install -y libasound2-dev  liblog4cpp5-dev liblivemedia-dev

Best Regards,
Michel.

regisin, 2016-09-29:
I have the same problem. Do we have to install any ALSA related library? I cloned the repository, and commented out the cmakefile lines where it tries to find the ALSA libs.

regisin, 2016-09-29:
Rebooted after installing the dependencies (libasound2-dev). Also, would recommend to install dependencies before cmake

ronnyvdbr, 2016-09-29:
Hi, thanks for the feedback, I started from scratch with the latest jessie minimal, I installed the given dependencies and it worked right out of the box then.

sudo apt-get install -y libasound2-dev liblog4cpp5-dev liblivemedia-dev
cd v4l2rtspserver
cmake . && make
sudo make install

pi@raspberrypi:~/v4l2rtspserver $ sudo make install
[100%] Built target v4l2rtspserver
Install the project...
-- Install configuration: "DEBUG"
-- Installing: /usr/local/bin/v4l2rtspserver
pi@raspberrypi:~/v4l2rtspserver $

closed 2016-10-09 by mpromonet

mlsomers, 2019-01-12:
Just for all others running into this error (google will get you to this issue):

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> V4L2WRAPPER_LIBRARY
>     linked by target "v4l2rtspserver" in directory /home/pi/v4l2rtspserver

You will get this if you downloaded the zip or tar.gz files instead of:
`sudo git clone https://github.com/mpromonet/v4l2rtspserver.git`

THat's because v4l2wrapper is a linked repository just like hls.js (which you will run into later if you decide to fix incrementally). I chose to do it this way and these are the steps I needed (I hope it's complete, left out allot of dead ends):

```
sudo mkdir v4l2wrapper
sudo git clone https://github.com/mpromonet/libv4l2cpp.git v4l2wrapper
rmdir /home/pi/v4l2rtspSrc/armv6l/include/libv4l2cpp
sudo make -C v4l2wrapper CC=/usr/bin/cc PREFIX=/home/pi/v4l2rtspSrc/armv6l all install 2>&1
sudo mkdir hls.js
sudo git clone https://github.com/video-dev/hls.js.git hls.js
cd hls.js
sudo mkdir dist
cd dist
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.js
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.js.map
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.light.js
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.light.js.map
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.light.min.js
sudo wget https://github.com/video-dev/hls.js/raw/e90a1b999071e88af0f734ef18b9e8520c839cec/dist/hls.min.js
cd ..
cd ..
sudo cmake .
sudo make
sudo make install
```
In hindsight I guess you could skip getting hls.js from git (it skips the 'dist' directory any way) and just wget `hls.js/dist/hls.light.min.js`. Hope that helps...