mpromonet/v4l2rtspserver #37: Cant start server

climax89, 2016-10-02:
Hey guys,

i can't start the server.

Webcam: Logitech C525

This is the message:

```
./v4l2rtspserver /dev/video0
2016-10-02 20:56:18,647 [NOTICE] - /home/patrick/v4l2rtspserver/src/main.cpp:352
	Create V4L2 Source.../dev/video0
2016-10-02 20:56:18,648 [NOTICE] - src/V4l2Device.cpp:115
	driver:uvcvideo 4000001
2016-10-02 20:56:18,649 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video0 support streaming
2016-10-02 20:56:18,649 [ERROR ] - src/V4l2Device.cpp:159
	Cannot set format for device:/dev/video0 Invalid argument
2016-10-02 20:56:18,650 [ERROR ] - src/V4l2Device.cpp:65
	Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
```

mpromonet, 2016-10-02:
Hi,
I guess your device doesnot support H264 format.
You could check that using `v4l2-ctl --list-formats`.
In order to compress it, you could use an intermediate virtual device.
Regards,
Michel.

climax89, 2016-10-02:
```
v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUV 4:2:2 (YUYV)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : MJPEG
```

mpromonet, 2016-10-03:
Hi,
Then your device doesnot support H264 capture.
A possible way that is describe in the [README.md](https://github.com/mpromonet/v4l2rtspserver/blob/master/README.md) involve [v4l2loopback](https://github.com/umlaeute/v4l2loopback) and [v4l2compress_h264](https://github.com/mpromonet/v4l2tools).
```
modprobe v4l2loopback video_nr=10
v4l2compress_h264 /dev/video0 /dev/video10 &
v4l2rtspserver /dev/video10 &
```
Regards,
Michel.

climax89, 2016-10-03:
Thank you very much! :)

closed 2016-10-09 by mpromonet

AndreV84, 2017-09-06:
what if the output is:
v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RGGB'
	Name        : 8-bit Bayer RGRG/GBGB

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'BG10'
	Name        : 10-bit Bayer BGBG/GRGR

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'RG12'
	Name        : 12-bit Bayer RGRG/GBGB

AndreV84, 2017-09-06:
Create V4L2 Source.../dev/video1
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

mpromonet, 2017-09-06:
Hi AndreV84,
The answer is the same as before, this format are not supported as an RTP payload by live555.
Then it needs some conversion to one of the format supported by this program (JPEG/MJPG/H264,VP8 and in experiment VP9/HEVC)
Best Regards,
Michel.

AndreV84, 2017-09-06:
hi mpromonet,
thank you for your response.
it seems that I shall install https://github.com/mpromonet/v4l2tools which seems to be a challenge returning errors...

AndreV84, 2017-09-06:
can that be used: video/x-raw,width=640,height=480,format=I420 ?

JShitote, 2017-09-07:
I had the same problem and solved it by adding the line:
bcm2835-v4l2 
to sudo vi /etc/modules

A good guide would be the following link 
https://forum.synology.com/enu/viewtopic.php?t=98870

Hope this helps :) :)