mpromonet/v4l2rtspserver #38: Information about latency

glerchundi, 2016-10-31:
First of all, this is an incredible work, thanks for sharing!

I would like to know how much is a reasonable value range for latency using this project? I'm seeing something around 450ms - 550ms using Raspberry Pi 2 in a WiFi network.

In case you have no benchmarking data, which would be the best way to measure it? My testing was really really awful with an stopwatch running (recording it with the RPi Camera) and a computer running a VLC client pointing to the RTSP server (where the image appears with <1s latency), everything recorded by another phone and then comparing them...

Thanks in advance,

mpromonet, 2016-11-06:
Hi Gorka,

A slightly simpler measurment method could be to display a milliseconds counter (like your stopwatch) on the same display used by the RTSP client.
Then pointing the camera on the screen in a way you could see the counter and the recorded counter, you get the latency substracting the 2 counter from a snapshot of the screen.

I made some tests using such a way, but I don't remember the latency values. Obviously this is highly dependent on the resolution, and also the player used.
VLC even with no network caching introduce an important latency, ffplay have less.

Best Regards,

Michel.

glerchundi, 2016-11-07:
> A slightly simpler measurment method could be to display a milliseconds counter (like your stopwatch) on the same display used by the RTSP client.
> Then pointing the camera on the screen in a way you [...]

yep :)

> I made some tests using such a way, but I don't remember the latency values. Obviously this is highly dependent on the resolution, and also the player used.
> VLC even with no network caching introduce an important latency, ffplay have less.

The tests I did last week were with:

* Using `ffmpeg -i rtsp://192.168.86.155:8554/unicast -f sdl -` (which is the "new" way of doing `ffplay` in OS X / macOS - deprecated `ffplay` in macOS is sloooow):

[image]

Aprox. **430ms**

* Using `VLC` (no network caching):

[image]

Aprox. **440ms**

Closing this then, thanks for your comment!

closed 2016-11-07 by glerchundi