mpromonet/v4l2rtspserver #39: VP9 support

saket424, 2016-11-25:
I bumped into this project just today. I verified VP8 works. Great work Michel !

sudo modprobe v4l2loopback video_nr=10
v4l2compress_vp8 /dev/video0 /dev/video10
v4l2rtspserver -fVP80 /dev/video10

What is involved to add VP9 support? I believe Live555 does support VP9

mpromonet, 2016-11-25:
Hi,
You are right live555 support VP9, but V4L2 kernel include seems to not define it [videodev2.h](https://github.com/torvalds/linux/blob/master/include/uapi/linux/videodev2.h) even if it is described in [V4L2 documentation](https://www.linuxtv.org/downloads/v4l-dvb-apis-new/uapi/v4l/pixfmt-013.html)
However it is possible to use directly v4l2_fourcc('V', 'P', '9', '0') or define V4L2_PIX_FMT_VP9.

I guess adding support of VP9 is straight forward : 
 - libvpx support VP9 then initialization of codec should be modified in https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2compress_vp8.cpp 
 - live555 support VP9 then initialization of  RTPSink should be added in src/ServerMediaSubsession.cpp and little modification in src/main.cpp

Best Regards,
Michel.

saket424, 2016-11-25:
Hi Michel,
I tried making the changes outlined by you yesterday itself and was not successful as the v4l2 kernel module did not recognize VP9 format even after I successfully compiled v4l2compress_vp9

Could you give it a try and see what the issue is ?

Thanks

mpromonet, 2016-11-26:
Hi,
You are right, v4l2loopback doesnot support VP9 [v4l2loopback_formats.h](https://github.com/umlaeute/v4l2loopback/blob/master/v4l2loopback_formats.h) , it needs to modify it.
I am making a try and will let you know.
Regards,
Michel

saket424, 2016-12-01:
I am still having issues with VP90. VP80 works but VP90 does not
I get an ERROR -- "Cannot set pixelformat to:VP90 format is:BGR4"

anand@saket9:~/rtsp-stuff/v4l2tools$ ./v4l2compress_vpx -f VP80 /dev/video0 /dev/video10
2016-12-01 13:04:00,008 [NOTICE] - src/V4l2Device.cpp:115
	driver:uvcvideo 84200001
2016-12-01 13:04:00,008 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video0 support streaming
2016-12-01 13:04:00,019 [NOTICE] - src/V4l2Device.cpp:177
	/dev/video0:YUYV size:640x480 bufferSize:614400
2016-12-01 13:04:00,029 [NOTICE] - src/V4l2Device.cpp:198
	fps:1/30
2016-12-01 13:04:00,029 [NOTICE] - src/V4l2Device.cpp:199
	nbBuffer:0
2016-12-01 13:04:00,030 [NOTICE] - src/V4l2MmapDevice.cpp:70
	Device /dev/video0 nb buffer:10
2016-12-01 13:04:00,264 [NOTICE] - src/V4l2Device.cpp:115
	driver:v4l2 loopback 85208002
2016-12-01 13:04:00,264 [NOTICE] - src/V4l2Device.cpp:117
	/dev/video10 support read/write
2016-12-01 13:04:00,264 [NOTICE] - src/V4l2Device.cpp:118
	/dev/video10 support output
2016-12-01 13:04:00,264 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video10 support streaming
2016-12-01 13:04:00,266 [NOTICE] - src/V4l2Device.cpp:177
	/dev/video10:VP80 size:640x480 bufferSize:1228800
2016-12-01 13:04:00,266 [NOTICE] - src/V4l2MmapDevice.cpp:70
	Device /dev/video10 nb buffer:8
2016-12-01 13:04:00,267 [NOTICE] - src/v4l2compress_vpx.cpp:186
	Start Capturing from /dev/video0
2016-12-01 13:04:00,267 [NOTICE] - src/v4l2compress_vpx.cpp:192
	Start Compressing /dev/video0 to /dev/video10

anand@saket9:~/rtsp-stuff/v4l2tools$ ./v4l2compress_vpx -f VP90 /dev/video0 /dev/video10
2016-12-01 13:05:36,616 [NOTICE] - src/V4l2Device.cpp:115
	driver:uvcvideo 84200001
2016-12-01 13:05:36,616 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video0 support streaming
2016-12-01 13:05:36,627 [NOTICE] - src/V4l2Device.cpp:177
	/dev/video0:YUYV size:640x480 bufferSize:614400
2016-12-01 13:05:36,637 [NOTICE] - src/V4l2Device.cpp:198
	fps:1/30
2016-12-01 13:05:36,637 [NOTICE] - src/V4l2Device.cpp:199
	nbBuffer:0
2016-12-01 13:05:36,637 [NOTICE] - src/V4l2MmapDevice.cpp:70
	Device /dev/video0 nb buffer:10
2016-12-01 13:05:36,876 [NOTICE] - src/V4l2Device.cpp:115
	driver:v4l2 loopback 85208002
2016-12-01 13:05:36,876 [NOTICE] - src/V4l2Device.cpp:117
	/dev/video10 support read/write
2016-12-01 13:05:36,876 [NOTICE] - src/V4l2Device.cpp:118
	/dev/video10 support output
2016-12-01 13:05:36,876 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video10 support streaming
2016-12-01 13:05:36,876 [ERROR ] - src/V4l2Device.cpp:164
	Cannot set pixelformat to:VP90 format is:BGR4
2016-12-01 13:05:36,877 [ERROR ] - src/V4l2Device.cpp:65
	Cannot init device:/dev/video10
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2016-12-01 13:05:36,877 [WARN  ] - src/v4l2compress_vpx.cpp:182
	Cannot create V4L2 output interface for device:/dev/video10

mpromonet, 2016-12-01:
Hi,
This is because v4l2loopback doesnot support VP9, you need to modify it, and reload de kernel module.
Doing this, I was able to run v4l2compress_vpx, however I was not able to read the VP9 using ffplay or vlc.
Best Regards,
Michel.

saket424, 2016-12-02:
Strange I still see the same error, I pulled v4l2loopback from your repository (git clone https://github.com/mpromonet/v4l2) and added these lines to v4l2loopback_formats.h and reloaded the v44l2loopback kernel modle
#ifdef V4L2_PIX_FMT_VP9
{
   .name     = "VP9",
     .fourcc   = V4L2_PIX_FMT_VP9,
     .depth    = 32,
     .flags    = FORMAT_FLAGS_COMPRESSED,
     },
#endif /* V4L2_PIX_FMT_VP9 */

Are there more v4l2loopback changes than the one above?

mpromonet, 2016-12-02:
Hi,
This deosnot work because V4L2_PIX_FMT_VP9 is not defined in kernel header.
Instead you could use : 
{
.name = "VP9",
.fourcc = v4l2_fourcc('V', 'P', '9', '0'),
.depth = 32,
.flags = FORMAT_FLAGS_COMPRESSED,
},
My fork of [v4l2loopback](https://github.com/mpromonet/v4l2loopback) is quite old, and I didnot commit my tries.
Regards,
Michel.

saket424, 2017-02-28:
I freshly clone v4l2loopback and added the following lines to v4l2loopback_formats.h
#define V4L2_PIX_FMT_VP9      v4l2_fourcc('V', 'P', '9', '0') /* VP9 */
#ifdef V4L2_PIX_FMT_VP9
{
   .name     = "VP9",
     .fourcc   = V4L2_PIX_FMT_VP9,
     .depth    = 32,
     .flags    = FORMAT_FLAGS_COMPRESSED,
     },
#endif /* V4L2_PIX_FMT_VP9 */

It works now. Thanks. Closing issue

closed 2017-02-28 by saket424