mpromonet/v4l2rtspserver #42: rtsp url 255.255.255.255:8554

scramble45, 2017-02-07:
Whenever I start using:
````
v4l2rtspserver -I /dev/video0 -P 8554
````

**I get the following:**
````
2017-02-07 13:56:00,029 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:386
        Create V4L2 Source.../dev/video0
2017-02-07 13:56:00,029 [NOTICE] - src/V4l2Device.cpp:115
        driver:bm2835 mmal 85200005
2017-02-07 13:56:00,029 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-02-07 13:56:00,030 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-02-07 13:56:00,031 [NOTICE] - src/V4l2Device.cpp:177
        /dev/video0:H264 size:640x480 bufferSize:307200
2017-02-07 13:56:00,031 [NOTICE] - src/V4l2Device.cpp:198
        fps:1/25
2017-02-07 13:56:00,031 [NOTICE] - src/V4l2Device.cpp:199
        nbBuffer:1
2017-02-07 13:56:00,035 [NOTICE] - src/V4l2MmapDevice.cpp:70
        Device /dev/video0 nb buffer:10
2017-02-07 13:56:00,574 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:407
        Create Source .../dev/video0
2017-02-07 13:56:00,574 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-02-07 13:56:00,576 [NOTICE] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
        profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0A8SJqA=,KO4fLA==
cannot set sample format (Invalid argument)
2017-02-07 13:56:00,582 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://255.255.255.255:8554/unicast"
2017-02-07 13:56:00,582 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-02-07 13:56:00,582 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:202
        V4L2DeviceSource::getNextFrame errno:0 Success
2017-02-07 13:56:00,582 [ERROR ] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
        error:Success
2017-02-07 13:56:00,582 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
        end thread

````

**When I don't specify the interface it gives me an actual IP:**

````
$ sudo v4l2rtspserver -P 8554
2017-02-07 13:56:49,450 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:386
        Create V4L2 Source.../dev/video0
2017-02-07 13:56:49,450 [NOTICE] - src/V4l2Device.cpp:115
        driver:bm2835 mmal 85200005
2017-02-07 13:56:49,451 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-02-07 13:56:49,451 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-02-07 13:56:49,452 [NOTICE] - src/V4l2Device.cpp:177
        /dev/video0:H264 size:640x480 bufferSize:307200
2017-02-07 13:56:49,453 [NOTICE] - src/V4l2Device.cpp:198
        fps:1/25
2017-02-07 13:56:49,453 [NOTICE] - src/V4l2Device.cpp:199
        nbBuffer:1
2017-02-07 13:56:49,456 [NOTICE] - src/V4l2MmapDevice.cpp:70
        Device /dev/video0 nb buffer:10
2017-02-07 13:56:49,995 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:407
        Create Source .../dev/video0
2017-02-07 13:56:49,996 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-02-07 13:56:49,996 [NOTICE] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
        profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0A8SJqA=,KO4fLA==
cannot set sample format (Invalid argument)
2017-02-07 13:56:50,003 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-02-07 13:56:50,003 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://192.168.251.114:8554/unicast"
2017-02-07 13:56:50,004 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:202
        V4L2DeviceSource::getNextFrame errno:0 Success
2017-02-07 13:56:50,004 [ERROR ] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
        error:Success
2017-02-07 13:56:50,004 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
        end thread
^CSIGINT
2017-02-07 13:56:51,861 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:505
        Exiting....
````

scramble45, 2017-02-07:
Also I have been trying to set a different interface using:
````
-I /dev/video1
````

It just seems to ignore that and try the default /dev/video0

mpromonet, 2017-02-07:
Hi scramble45,

The option `-I` is used to specify the output network interface.
If you would like to feed from /dev/video1, you could use :  

    v4l2rtspserver /dev/video1

If you would like to spefify the restrict the RTSP server to local client, you could use 

    v4l2rtspserver -I 127.0.0.1

Regards,
Michel.

scramble45, 2017-02-07:
Guess I didn't read the flags properly. Thanks for clearing that up. I was able to get both cameras going.

closed 2017-02-07 by scramble45

DragonDraco, 2017-04-18:
Hello ,

I managed to make the stream working , however just lost it and I am getting exactly same issue . 
Would you please let me know how did you solve it ?thank you in advance

mpromonet, 2017-04-19:
Hi,
To solve the issue you should use -I option to specify the network interface to listen to.
Regards,
Michel