mpromonet/v4l2rtspserver #43: Installation problem

LeonardoGentile, 2017-02-08:
Hello,
I've followed the instruction for building the server on a rpi and it seems to work when the generated bin is launched from within the directory where it was compiled. 

I've tried to install it with `sudo make install` and the bin was copied to `/usr/local/bin/v4l2rtspserver` but when referencing this binary either explicity (full path) or just invoking the bin name it seems to look for the source directory (the source directory from where it was compiled). I think this not the intended behaviour, right? Even if I keep my source folder there it seems to throw quite some errors.

By the way to re-try the whole installation process how am I supposed to cleanup everything that was installed and re-try? Just removing `/usr/local/bin/v4l2rtspserver` would do it?

mpromonet, 2017-02-13:
Hi,
I am not sure to understand 

>  it seems to look for the source directory (the source directory from where it was compiled)

The log trace reference the source path, but this doesnot impact the runtime.
Could you clarify your issue ?
Regards,
Michel.

LeonardoGentile, 2017-02-13:
Ah, I see, so it's only the log trace referencing the source path. So I guess removing the src folder won't throw any error, right?

Last question, how to completely remove the installed bin (and possibly any installed libraries)? Just a simple `/usr/local/bin/v4l2rtspserver`?

mpromonet, 2017-02-13:
Yes, you can remove the src folder.
and yes to uninstall you just need to remove the executable, there is no installed librairy.

closed 2017-02-13 by mpromonet