mpromonet/v4l2rtspserver #45: Any option to specify target bitrate?

saket424, 2017-02-13:
I currently don't see an option to specify a target bitrate. Any ideas on how the bitrate constraints can be honored?

mpromonet, 2017-02-13:
Hi,

This program read from V4L2 and write to RTP, the bitrate depends on V4L2 device.
Then the possibility to control it depends on V4L2 driver you are using.

For instance on raspberry, you can choose constant bitrate or variable bitrate with : 

   v4l2-ctl -c video_bitrate_mode=0 (1 for constant)

and bitrate using : 

   v4l2-ctl -c video_bitrate=10000000

Regards,
Michel.

saket424, 2017-02-28:
I see that there is a difference between the command line options in v4l2compress_vpx and v4l2compress_h264. One allows the target bitrate to be specified and the other does not. This may be an issue for v4l2tools instead of v4l2rtspserver

$ ./v4l2compress_vpx --help
./v4l2compress_vpx [-v[v]] [-W width] [-H height] source_device dest_device
	 -v                   : verbose 
	 -vv                  : very verbose 
	 -b bitrate           : target bitrate 
	 -c                   : rate control mode CBR (default is VBR) 
	 -f format            : format (default is VP80) 
	 -W width             : V4L2 capture width (default 640)
	 -H height            : V4L2 capture height (default 480)
	 -F fps               : V4L2 capture framerate (default 25)
	 -r                   : V4L2 capture using read interface (default use memory mapped buffers)
	 -w                   : V4L2 capture using write interface (default use memory mapped buffers)
	 source_device        : V4L2 capture device (default /dev/video0)
	 dest_device          : V4L2 capture device (default /dev/video1)

$ ./v4l2compress_h264 --help
./v4l2compress_h264 [-v[v]] [-W width] [-H height] source_device dest_device
	 -v            : verbose 
	 -vv           : very verbose 
	 -W width      : V4L2 capture width (default 640)
	 -H height     : V4L2 capture height (default 480)
	 -F fps        : V4L2 capture framerate (default 10)
	 -r            : V4L2 capture using read interface (default use memory mapped buffers)
	 -w            : V4L2 capture using write interface (default use memory mapped buffers)
	 source_device : V4L2 capture device (default /dev/video0)
	 dest_device   : V4L2 capture device (default /dev/video1)

closed 2017-03-09 by mpromonet

linng85, 2017-07-14:
Hi i tried the **v4l2-ctl -2 video_bitrate=10000000** and **v4l2-ctl --set-ctrl video_bitrate=1000000**. It return the result unknown control 'video_bitrate'. Any idea on the problem?

mpromonet, 2017-07-16:
Hi inng85
Probaly the control 'video_bitrate' is not supported by the driver you are using.
You can check with `v4l2-ctl -all`
Best Regards,
Michel.

linng85, 2017-07-17:
Thanks a lot Michel! You are right it is not in the list of the driver. Means that i need to reinstall the driver?

mpromonet, 2017-07-18:
Hi linng85,
You can try to  reinstall the driver, but it may have no effect.
It's probably means that the driver does not have this control, maybe a similar one exists.
Regards,
Michel.