mpromonet/v4l2rtspserver #46: Feature Request: Add on/off options for audio/video streams

bdheeman, 2017-02-20:
Sometime, we don't have audio device, e.g. when using cheap USB webcams; or it could be a project requirement not to send an audio stream even if available as well.  So we need options for ignoring non-availability of audio devices; currently the program aborts if the audio device is not available.

Similarly, sometimes we don't want to stream the video; also depends of requirements of a project and, or non availability of video device ;-)

mpromonet, 2017-02-20:
Hi,

I think you can stream only audio using : 

            v4l2rtspserver /dev/viideo0,

In the same idea you can stream only audio using : 

           v4l2rtspserver ,default

Regards,
Michel.

mpromonet, 2017-03-12:
Hi,
I add some explaination in the commit 12d876adc084b59a3c698ab9ccd8a2073bd3e956
Best Regards,
Michel.

closed 2017-03-12 by mpromonet