mpromonet/v4l2rtspserver #50: Pi Audio

MichaelAronsen, 2017-05-08:
Hi,

Should this work on a Pi 3 with Pi camera module and using a ps eye webcam with microphone array for sound?

Microphone array is configured as a single device and as the default alsa input source..

mpromonet, 2017-05-08:
Hi,
If the ALSA device support SND_PCM_FORMAT_S16_BE, SND_PCM_ACCESS_RW_INTERLEAVED with 2 channels, it should works.
If you're using other options, you should modify the ALSACapture.
Regards,
Michel.

MichaelAronsen, 2017-05-08:
Hi,

Thanks for answering this, unfortunately i'm quite inexperienced with alsa, how would I check if it supports these?

Thanks,
Michael

mpromonet, 2017-05-10:
Hi,
Maybe you could check if it support this configuration, trying it ?
Regards,
Michel.

MichaelAronsen, 2017-05-11:
Hi,

I did try it out, even changed ALSA params to 4 channels as the ps3 eye has a 4 channel microphone array, but all i'm getting is static noise.

I've matched to bitrate to what arecord -v reports the device provides, but no go..

If it helps and you're interested, i'd be happy to order one of these ps3 eye cameras on amazon.fr for you..

mpromonet, 2017-05-13:
Hi Michael,
Your problem seems similar to #51 ?
Only S16_BE was supported, because I would like to avoid audio conversion.
In the commit 613ece8895354ca689a9c9281662d8d2016eda65, I add the possibility to specify the capture audio format (S16_LE & S16_BE) and in a632b60793cd7ee27852239127b88e8ed6b94373 I add more format (S32_BE & S32_LE).
Maybe it solve your problem ?
Best Regards,
Michel.

MichaelAronsen, 2017-05-16:
Hi Michel,
Thanks, it's definitely a step in the right direction, but as mentioned in #51, the audio captured is noise, no actual audio.

 # cat /proc/asound/card1/stream0 
OmniVision Technologies, Inc. USB Camera-B4.09.24.1 at usb-3f980000.usb-1.4, hi : USB Audio

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S16_LE
    Channels: 4
    Endpoint: 4 IN (ASYNC)
    Rates: 16000
    Data packet interval: 1000 us

So, I do need to use S16_LE format, with -C 4 and -A 16000.

Just not sure why i'm getting a rhythmic static noise, rather than the actual audio in the room.

MichaelAronsen, 2017-05-16:
Sample here:
http://aronsen.cloud.tilaa.com/michael/test.mp4

mpromonet, 2017-07-23:
Hi Michael,
This may be fixed by pull request https://github.com/mpromonet/v4l2rtspserver/pull/55
If not, you could reopen a new issue.
Best Regards,
Michel.

closed 2017-07-23 by mpromonet