mpromonet/v4l2rtspserver #51: Record Audio from sound device

masoudr, 2017-05-09:
Hi,
I'm using this method to stream both audio and video. for the video part, it works fine but I can't get it working with my USB sound card. The Alsa configuration is just fine and I can record with the following command:
 `arecord --device=hw:1,0 --format S16_LE --rate 44100 -c1 test.wav`
but I can't get it working with rtspserver, I use the following command but I only get video without any sound.
`v4l2rtspserver -W 1920 -H 1080 /dev/video0,1`
Thanks.

mpromonet, 2017-05-10:
Hi,
I guess you should try with : 
      v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1
If the name of the alsa device need to be `hw:1,0`, the parser should be updated.
Best Regards,
Michel.

mpromonet, 2017-05-12:
Hi,

The problem occurs problably with `cannot set sample format (Invalid argument)`.
This correspond to  : 

            snd_pcm_hw_params_set_format (m_pcm, hw_params, SND_PCM_FORMAT_S16_BE)

You can make some tries with other format.
I may add an option to configure this through the command line.
Best Regards,
Michel.

masoudr, 2017-05-12:
What do you mean by "tries with other format".
How can I change the format? As I know, I tested this sound card with other programs, it can work with a sample rate of 48000 Hz and mp3 output format. Can you explain a little more specific about that? Which file should I edit before compiling the program?
Thank you.

mpromonet, 2017-05-13:
Hi,
It seems your capture device support only Little Endian, but the network order in Big Endian, I commit an ugly way to convert S16_LE to S16_BE.
Maybe, you will be able to capture audio adding : 

    -a S16_LE

Best Regards,
Michel.

masoudr, 2017-05-14:
You were right, it only supports `S16_LE` as you can see the output of `cat /proc/asound/card1/stream0`:
`pi@raspberrypi:~/rtsp_stream$ cat /proc/asound/card1/stream0`
```
C-Media Electro Inc. USB PnP Sound De at usb-20980000.usb-1.1, full speed : USB Audio
Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 1 OUT (ADAPTIVE)
    Rates: 48000, 44100
Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S16_LE
    Channels: 1
    Endpoint: 2 IN (ADAPTIVE)
    Rates: 48000, 44100
```
I used this command to run:
`v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -C 1 -a S16_LE -A 48000`
The sound came out but it was only some noise I couldn't hear any of my said words in it.

masoudr, 2017-05-20:
I tried the new version, now I can hear my voice but still, it is not good. In every second it is getting muted. I think the problem is with the buffer size, hope this bug fix soon and we can have an absolutely perfect video streamer.
Thank you.

RonnyStrutz, 2017-05-25:
Hi,

there is a bug in your v4l2rtspserver, probably introduced with the endian reordering stuff. Pls check your ssize_t read(){} in ALSACapture.h, snd_pcm_readi() returns the number of frames, not bytes. This causes the strange sounds which users reported here.

My attached patch will fix this.

Best
Ronny

[fix.txt](https://github.com/mpromonet/v4l2rtspserver/files/1029816/fix.txt)

mpromonet, 2017-05-25:
Hi Ronny,
Your patch seems good.
I think it is better to initialize fmt_phys_width_bytes to 0.
It could be nice that you submit a pull request with your contribution, I will then merge it.
Thanks & Regards,
Michel.

closed 2017-07-23 by mpromonet