mpromonet/v4l2rtspserver #52: Static image / grey

PerryVanKuppeveld, 2017-05-18:
I am using the v4l2rtspserver with the v4l2loopback on a Raspberry Pi 3, with a normal usb webcam (no raspi cam).
Creation of the virtual loopback adapter works fine. Using the v4l2compress_h264 to feed the virtual device.
v4l2display_h264 shows the video from the virtual device. So far so good.

I can start the v4l2rtspserver. When I view the generated stream using VLC (on an other computer) I see a static image (from the camera) for one second, then a grey filled square.

Any ideas what can be wrong? do you need som traces?

mpromonet, 2017-05-23:
Hi,
Maybe network is loosing too much frames.
Did you try asking for stream over tcp ?
Regards
Michel

PerryVanKuppeveld, 2017-05-24:
I think VLC is the problem. Watching the stream in the ONVIF Device Manager shows the proper stream.

Using the v4l2rtspserver with a raspicam works perfect, great result at no cpu usage! Perfect!
Now I need to write the rtsp stream back to the v4l2 loopback virtual device. Any suggestions?

closed 2017-05-24 by PerryVanKuppeveld

mpromonet, 2017-05-25:
Hi,
To write an H264 RTSP stream to a v4l2 device, you can look at https://github.com/mpromonet/v4l2rtspclient, maybe it could helps you ?
Best Regards,
Michel.

PerryVanKuppeveld, 2017-05-26:
Oh, wow, how could I have missed this. Gonna try it. Thanks! By the way,  great stuff you have made.