mpromonet/v4l2rtspserver #53: Can't view mjpeg stream

EasonLLL, 2017-05-19:
Hello ,
 When i start  the v4l2rtspserver , vlc doesn't play anything.

- image with VLC
[image]

- This is the message:
> root@mylinkit:/#
>  `./v4l2rtspserver -fMJPG -P 8554 /dev/video0  -s -v -W 640 -H 480 -F 25`
> 2017-05-17 13:41:45,180 [INFO  ] - /media/sf_TFTP/STREAM_SERVER/V4L2RTSPSERVER_openwrt_wooya/STREAMSERVER/TARGET/include/logger.h:44
>         level:INFO
> 2017-05-17 13:41:45,183 [NOTICE] - src/main.cpp:392
>         Create V4L2 Source.../dev/video0
> 2017-05-17 13:41:45,184 [NOTICE] - src/V4l2Device.cpp:115
>         driver:uvcvideo 84000001
> 2017-05-17 13:41:45,184 [NOTICE] - src/V4l2Device.cpp:119
>         /dev/video0 support streaming
> 2017-05-17 13:41:45,190 [NOTICE] - src/V4l2Device.cpp:177
>         /dev/video0:MJPG size:640x480 bufferSize:921600
> 2017-05-17 13:41:45,195 [NOTICE] - src/V4l2Device.cpp:198
>         fps:1/25
> 2017-05-17 13:41:45,196 [NOTICE] - src/V4l2Device.cpp:199
>         nbBuffer:0
> 2017-05-17 13:41:45,215 [NOTICE] - src/V4l2MmapDevice.cpp:70
>         Device /dev/video0 nb buffer:10
> 2017-05-17 13:41:45,216 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:0 size:921600
> 2017-05-17 13:41:45,217 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:1 size:921600
> 2017-05-17 13:41:45,218 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:2 size:921600
> 2017-05-17 13:41:45,218 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:3 size:921600
> 2017-05-17 13:41:45,219 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:4 size:921600
> 2017-05-17 13:41:45,220 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:5 size:921600
> 2017-05-17 13:41:45,221 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:6 size:921600
> 2017-05-17 13:41:45,222 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:7 size:921600
> 2017-05-17 13:41:45,223 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:8 size:921600
> 2017-05-17 13:41:45,223 [INFO  ] - src/V4l2MmapDevice.cpp:89
>         Device /dev/video0 buffer idx:9 size:921600
> 2017-05-17 13:41:45,226 [NOTICE] - src/main.cpp:413
>         Create Source .../dev/video0
> 2017-05-17 13:41:45,227 [NOTICE] - src/main.cpp:171
>         getRtpFormat=video/JPEG
> 2017-05-17 13:41:45,227 [NOTICE] - src/main.cpp:147
>         Play this stream using the URL "rtsp://10.10.10.254:8554/unicast"
> 2017-05-17 13:41:46,431 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028506 fps:1 bandwidth:545kbps
> 2017-05-17 13:41:47,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028507 fps:15 bandwidth:8181kbps
> 2017-05-17 13:41:48,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028508 fps:24 bandwidth:13089kbps
> 2017-05-17 13:41:49,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028509 fps:24 bandwidth:13102kbps
> 2017-05-17 13:41:50,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028510 fps:24 bandwidth:13086kbps
> 2017-05-17 13:41:50,299 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         outtv_sec:1495028510 fps:1 bandwidth:546kbps
> 2017-05-17 13:41:50,300 [INFO  ] - inc/MJPEGVideoSource.h:53
>         width:80 height:60 type:0
> 2017-05-17 13:41:50,301 [NOTICE] - inc/MJPEGVideoSource.h:67
>         Quantization table idx:0 size:128 total size:128
> 2017-05-17 13:41:50,317 [INFO  ] - inc/MJPEGVideoSource.h:53
>         width:80 height:60 type:0
> 2017-05-17 13:41:50,330 [INFO  ] - inc/MJPEGVideoSource.h:53
[... 116 lines cut ...]
>         V4L2DeviceSource::doStopGettingFrames
> 2017-05-17 13:41:52,209 [NOTICE] - src/V4l2DeviceSource.cpp:131
>         V4L2DeviceSource::doStopGettingFrames
> 2017-05-17 13:41:53,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028513 fps:24 bandwidth:13111kbps
> 2017-05-17 13:41:54,031 [INFO  ] - src/V4l2DeviceSource.cpp:29
>         intv_sec:1495028514 fps:24 bandwidth:13129kbps
> ^CSIGINT
> 2017-05-17 13:41:54,434 [NOTICE] - src/main.cpp:511
>         Exiting....

- My usb camera support format list

> root@mylinkit:/# v4l2-ctl --list-formats
> [14284.020000] uvcvideo: Failed to query (GET_DEF) UVC control 13 on unit 1: -32 (exp. 8).
> ioctl: VIDIOC_ENUM_FMT
>         Index       : 0
>         Type        : Video Capture
>         Pixel Format: 'MJPG' (compressed)
>         Name        : MJPEG
>         Index       : 1
>         Type        : Video Capture
>         Pixel Format: 'YUYV'
>         Name        : YUV 4:2:2 (YUYV)
>

mpromonet, 2017-05-23:
Hi,
The streamer handle format with fourcc JPEG, but not MJPG. 
Supporting MJPG need some conversion to fit an rtp payload format
Best regards
Michel

mpromonet, 2017-05-25:
Hi,
I think it should work with `-f JPEG`, but this will not work with your usb camera as it supports MJPG and YUYV
Regards,
Michel

mpromonet, 2017-06-04:
Hi,
I modify the code to process MJPG V4L2 format using video/JPEG RTP payload in this commit https://github.com/mpromonet/v4l2rtspserver/commit/33ae6101289641f0895137bd7a6b340c8ab809b2.
I guess you did ?
Regards,
Michel.

mpromonet, 2017-06-11:
Hi,
you should try with the last release, maybe it is working better ?
What is uncommon in the log you post is : 

> 2017-05-17 13:41:50,301 [NOTICE] - inc/MJPEGVideoSource.h:67
> Quantization table idx:0 size:128 total size:128

Usually there is 2 quantization table (one for Y and one for U & V) of 64 bytes.
I fix some error parsing the DQT frame that could change the behaviour.
Reagrds,
Michel.

mpromonet, 2017-07-23:
Hi,
I close this issue, as there were significant improvement on the subject.
If you still meet problem, you could raise a new one.
Regards,
Michel.

closed 2017-07-23 by mpromonet