mpromonet/v4l2rtspserver #54: Port to another board

masoudr, 2017-05-25:
Hi,
I was wondering if you can port this tool for another board other than raspberry pi, I tried to compile this tool for Friendly arm's Nano pi neo air (Ubuntu) but with no success.
Is it any chance to use this awesome tool with this board?
Thanks.

mpromonet, 2017-05-25:
Hi,
In the commit https://github.com/mpromonet/v4l2rtspserver/commit/ce6245dfc612cc56b256eead9a82b0babe7210c1, I added configuration for nanopi.
In order to build it, you should add live555 and log4cpp in your cross-compiler sysroot : 
```
wget https://kent.dl.sourceforge.net/project/log4cpp/log4cpp-1.1.x%20%28new%29/log4cpp-1.1/log4cpp-1.1.2.tar.gz -O - | tar xvzf -
pushd log4cpp
CC=arm-linux-gcc CXX=arm-linux-g++ ./configure --prefix=$(arm-linux-gcc -print-sysroot)/usr/ --host=arm
make 
sudo make install
popd

wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz -O - | tar xzf -
pushd live
./genMakefiles linux
make C_COMPILER=arm-linux-gcc CPLUSPLUS_COMPILER=arm-linux-g++ LINK='arm-linux-g++ -o' 
sudo make PREFIX=$(arm-linux-gcc -print-sysroot)/usr/ install
popd
```
Then you should be able to crosscompile using : 
```
cmake -DCMAKE_TOOLCHAIN_FILE=nanopi.toolchain .
make
```
Best Regards,
Michel.

masoudr, 2017-05-25:
Thanks.
Can you explain a little simple?
What do you mean by "cross compile sys root"?
You mean I need to first run these commands on my Linux pc right?
I am not a Linux guy can you make it clear?

mpromonet, 2017-05-25:
Hi,
When you install the cross-compiler [Setup_Cross_Compiler_for_User_Space_Programs](http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO#Setup_Cross_Compiler_for_User_Space_Programs), it embeds include and library in a folder that can be got using `gcc -print-sysroot`.
In order to add additionnals libraries (log4cpp & live555) in it you could run the command I gave in previous post.
Then using the cmake toolchain configuration I just add, you can build it.
I have no nanopi, so I cannot test it, but I build it without difficulty.
Best Regards,
Michel.

masoudr, 2017-05-26:
I am using the ready Ubuntu image which has drivers. Is there any chance to compile the code in the board itself?
I think it would be easier.

mpromonet, 2017-05-26:
Hi,
If your are using the target, it could work simply with : 

    sudo apt-get install -y cmake g++ liblog4cpp5-dev liblivemedia-dev libasound2-dev
    cmake . && make

Is it what you tried ?
Regards,
Michel.

masoudr, 2017-05-26:
I will try it and report it here.

mpromonet, 2017-05-26:
I don't know difference between rasberrypi and nanopi, maybe you can use the raspberry build https://github.com/mpromonet/v4l2rtspserver/releases/download/v0.0.2/v4l2rtspserver-0.0.2-Linux-armv7l.deb ?

masoudr, 2017-05-26:
The difference is only with the tools that are already made for Pi and of course the distribution on that platform is Ubuntu but most of the tools that made for cam streaming like RPiCamInterface doesn't compile due to using the engines like raspivid which is unavailable on nano pi, so I think there is no chance to use them, but your tool is better because it uses the v4l2 driver which is generally available so I think I can make it. I will use both of solution and report the results here in a few hours.
Thank you.

masoudr, 2017-05-27:
I tried to compile with above commands and it was compiled successfully. but when I want to start I get this error:
```
root@FriendlyARM:~/v4l2rtspserver# v4l2rtspserver /dev/video0
2017-05-27 02:34:31,848 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:393
        Create V4L2 Source.../dev/video0
2017-05-27 02:34:32,166 [NOTICE][  123.671449] [VFE_ERR]try bayer bus error when pix fmt is bayer rgb at try_fmt_internal!
 - src/V4l2Device.cpp:115
        driv[  123.682894] [VFE_ERR]pixel format (0x34363248) width 640 height 480 invalid at vidioc_s_fmt_vid_cap.
er:sunxi-vfe 5000001
2017-05-27 02:34:32,166 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-05-27 02:34:32,166 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-05-27 02:34:32,191 [ERROR ] - src/V4l2Device.cpp:159
        Cannot set format for device:/dev/video0 Invalid argument
2017-05-27 02:34:32,216 [ERROR ] - src/V4l2Device.cpp:65
        Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2017-05-27 02:34:32,217 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://127.0.1.1:8554/unicast"
```

mpromonet, 2017-05-27:
Hi,
This means that probably your V4L2 device doesnot support H264 capture.
Maybe you can look to https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2compress_h264.cpp to compress the video to a v4l2loopback device.
Regards,
Michel.

masoudr, 2017-05-27:
Yes, you are right, how can I use other formats like JPEG? (I know it can support it)
I tried to use this command but getting this output:
```
root@FriendlyARM:~/v4l2rtspserver# v4l2rtspserver -f JPEG /dev/video0
2017-05-27 06:34:52,942 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:393
        Create V4L2 Source...JPEG
VIDIOC_REQBUFS: Inappropriate ioctl for device
2017-05-27 06:34:52,943 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:414
        Create Source ...JPEG
2017-05-27 06:34:52,945 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://127.0.1.1:8554/JPEG/unicast"
2017-05-27 06:34:52,945 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-05-27 06:34:52,945 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:393
        Create V4L2 Source.../dev/video0
2017-05-27 06:34:52,945 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:206
        V4L2DeviceSource::getNextFrame no data errno:0 Success
2017-05-27 06:34:52,946 [ERROR ] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
        error:Success
2017-05-27 06:34:52,946 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
        end thread
2017-05-27 06:34:53,263 [NOTICE][ 1315.461227] [VFE_ERR]try bayer bus error when pix fmt is bayer rgb at try_fmt_internal!
 - src/V4l2Device.cpp:115
        driv[ 1315.472693] [VFE_ERR]pixel format (0x00000000) width 0 height 0 invalid at vidioc_s_fmt_vid_cap.
er:sunxi-vfe 5000001
2017-05-27 06:34:53,263 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-05-27 06:34:53,264 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-05-27 06:34:53,288 [ERROR ] - src/V4l2Device.cpp:159
        Cannot set format for device:/dev/video0 Invalid argument
2017-05-27 06:34:53,313 [ERROR ] - src/V4l2Device.cpp:65
        Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2017-05-27 06:34:53,314 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://127.0.1.1:8554/video0/unicast
```
The last line is not clear, I tried to use that url and replace the "127.0.1.1" with the ip op the board itself but it is not reachable.
About using v4l2loopback can you explain a little more how to use it?
Thanks.

mpromonet, 2017-05-29:
Hi,
You should not have a space between `-f` and the format using `-f JPEG /dev/video0`, try to open one device called JPEG and one /dev/video0.
You should try `-fJPEG /dev/video0`.
For the IP address, this is probably because the multicast is not routed to your IP interface (live555 use this way to detect the interface to use), and your hostname resolution is set to 127.0.1.1.
Regards,
Michel.

masoudr, 2017-05-30:
I tried the new version with above command and I got a new error about opening the device:
```
root@NanoPi-NEO-Air:/home/pi/v4l2rtspserver# ./v4l2rtspserver -fJPEG /dev/video0
2017-05-30 02:48:34,545 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:436
        Create V4L2 Source.../dev/video0
2017-05-30 02:48:34,855 [NOTICE] - src/V4l2Device.cpp:115
        driver:sunxi-vfe 5000001
2017-05-30 02:48:34,855 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-05-30 02:[  631.007745] [VFE_ERR]try bayer bus error when pix fmt is bayer rgb at try_fmt_internal!
48:34,856 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
[  631.033101] [VFE_ERR]pixel format (0x4745504a) width 640 height 480 invalid at vidioc_s_fmt_vid_cap.
2017-05-30 02:48:34,906 [ERROR ] - src/V4l2Device.cpp:159
        Cannot set format for device:/dev/video0 Invalid argument
2017-05-30 02:48:34,931 [ERROR ] - src/V4l2Device.cpp:65
        Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

```
then I tried above command with previous version and I got this:
```
root@FriendlyARM:~/v4l2rtspserver# v4l2rtspserver -f JPEG /dev/video0
2017-05-30 02:50:39,822 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:393
        Create V4L2 Source...JPEG
VIDIOC_REQBUFS: Inappropriate ioctl for device
2017-05-30 02:50:39,824 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:414
        Create Source ...JPEG
2017-05-30 02:50:39,824 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-05-30 02:50:39,824 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://127.0.1.1:8554/JPEG/unicast"
2017-05-30 02:50:39,824 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:206
        V4L2DeviceSource::getNextFrame no data errno:0 Success
2017-05-30 02:50:39,824 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:393
        Create V4L2 Source.../dev/video0
2017-05-30 02:50:39,825 [ERROR ] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
        error:Success
2017-05-30 02:50:39,825 [NOTICE] - /root/rstp/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
        end thread
2017-05-30 02:50:40,152 [NOTICE][   45.340941] [VFE_ERR]try bayer bus error when pix fmt is bayer rgb at try_fmt_internal!
 - src/V4l2Device.cpp:115
        driv[   45.352411] [VFE_ERR]pixel format (0x00000000) width 0 height 0 invalid at vidioc_s_fmt_vid_cap.
er:sunxi-vfe 5000001
2017-05-30 02:50:40,153 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2017-05-30 02:50:40,153 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-05-30 02:50:40,177 [ERROR ] - src/V4l2Device.cpp:159
        Cannot set format for device:/dev/video0 Invalid argument
2017-05-30 02:50:40,202 [ERROR ] - src/V4l2Device.cpp:65
        Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2017-05-30 02:50:40,202 [NOTICE] - /root/rstp/v4l2rtspserver/src/main.cpp:143
        Play this stream using the URL "rtsp://127.0.1.1:8554/video0/unicast"

```
it seems that the second part works right but how can I access the stream from network ?
I failed to access it with `rtsp://<ip add of NanoPi>:8554/video0/unicast`. How can I fix the routed IP with live555 ?
Thanks.

mpromonet, 2017-05-30:
Hi,
It seems to says that 640x480 at 25fps is not supported which is the dafault configuration.
You should check supported format with `v4l2-ctl --list-formats-ext` and select it with options 'H,W,F'.

Regards,
Michel.

masoudr, 2017-05-31:
I tried with other resolutions and fps but still, the same error here is the output of `v4l2-ctl --list-formats-ext`
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: '422P'
	Name        : planar YUV 422
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YU12'
	Name        : planar YUV 420
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'YV12'
	Name        : planar YVU 420
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'NV16'
	Name        : planar YUV 422 UV combined
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480

	Index       : 4
	Type        : Video Capture
	Pixel Format: 'NV12'
	Name        : planar YUV 420 UV combined
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
[... 247 lines cut ...]
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480

	Index       : 23
	Type        : Video Capture
	Pixel Format: 'BA12'
	Name        : RAW Bayer RGGB 12bit
		Size: Discrete 2592x1936
		Size: Discrete 2048x1536
		Size: Discrete 1920x1080
		Size: Discrete 1600x1200
		Size: Discrete 1280x960
		Size: Discrete 1280x720
		Size: Discrete 1024x768
		Size: Discrete 800x600
		Size: Discrete 640x480
```
I tried the v4l2loopback but due to missing kernel modules, it cannot be compiled.
so is there any way I can use this tool or I need to forget it ?
Thank you.

mpromonet, 2017-05-31:
Hi,
None of these format are compressed, and no uncompress format is supported by live555, the it needs something in the middle to compress it.
It is strange v4l2lookback build fails, may be you need to install some package like  linux-kernel-headers.
Next to compress it ffmpeg could be a way or some tools from https://github.com/mpromonet/v4l2tools.
I can easy write a JPEG compressor using it, or maybe omx is avaliable on nanopi ? I will probably order one soon, this board seems interesting.
Best Regards,
Michel.

masoudr, 2017-06-01:
Currently, I'm using FFmpeg to stream but I tested your tool with Pi zero and I think it was better.
I haven't tested omxplayer on this board but I think it would be work fine.
so I will wait until your next release.
Thanks for taking your time.

ykhandler, 2017-11-23:
hi. i've cross compile for mips. But when i run it, i've got this result:
./v4l2rtspserver /dev/video0
1970-01-01 08:02:19,680 [NOTICE] - /test/v4l2rtspserver/src/main.cpp:548
Create V4L2 Source.../dev/video0
1970-01-01 08:02:19,681 [ERROR ] - src/V4l2Device.cpp:82
Cannot open device:/dev/video0 Operation not permitted
1970-01-01 08:02:19,682 [ERROR ] - src/V4l2Device.cpp:65
Cannot init device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

can you tell me, what am i doing wrong? Thanks.

mpromonet, 2017-11-26:
@ykhandler 
It seems /dev/video0 cannot be openned by the user running v4l2rtspserver.
Maybe some command like `sudo chmod a+rw /dev/video0` or putting the user in the group that /dev/video0 belong could help.
Best Regards,
Michel.

closed 2018-01-14 by mpromonet