mpromonet/v4l2rtspserver #56: Output to ffmpeg via -O doesn't work?

vincentneo, 2017-05-28:
I used v4l2loopback to create `/dev/video1`,
then used `v4l2rtspserver -W 1920 -H 1080 -O /dev/video1 -S /dev/video0` 
after that `ffmpeg -i /dev/video1 -vcodec h264 -an -f segment -strftime 1 -segment_time 60 -segment_format mp4 " %d-%m_%H-%M-%S.mp4"`

But received only `no frame!` errors from ffmpeg. I think the v4l2rtspserver -O isn't working is it?
Anyway to fix it, or anyway to capture the stream, in the same way ffmpeg can do?

mpromonet, 2017-05-29:
Hi,
The -O feature was for debug, It writes what was read, then if your device never repeat the SPS & PPS, the virtual device will never receive it.
If you are using the official V4L2 for rapicam, you could enable this with : 

     v4l2-ctl  -c repeat_sequence_header=1

Best Regards,
Michel.

vincentneo, 2017-05-30:
` v4l2-ctl  -c repeat_sequence_header=1` works
but occasionally, I would get
` [h264 @ 0x2861380] decode_slice_header error`
`[h264 @ 0x2861380] no frame!`
at the start of ffmpeg still.
Restart the ffmpeg process and everything works again.

vincentneo, 2017-05-30:
I am also using `-vcodec copy` now instead by the way.

vincentneo, 2017-05-31:
Nevermind. Seems to work well. Thank You!! I'll close this issue now.

vincentneo, 2017-06-17:
Occasionally when starting FFmpeg for recording v4l2, this happens:
```PTS 1497684166626021, next:5866608 invalid dropping st:0
DTS 1497684166665288, next:5933274 st:0 invalid dropping
PTS 1497684166665288, next:5933274 invalid dropping st:0
DTS 1497684166705617, next:5999940 st:0 invalid dropping
PTS 1497684166705617, next:5999940 invalid dropping st:0
DTS 1497684166746298, next:6066606 st:0 invalid dropping
PTS 1497684166746298, next:6066606 invalid dropping st:0
DTS 1497684166786121, next:6133272 st:0 invalid dropping
PTS 1497684166786121, next:6133272 invalid dropping st:0
```
after Ctrl-c, this message appears
`Assertion av_rescale_rnd(start_dts, 1000, track->timescale, AV_ROUND_DOWN) <= 0 failed at libavformat/movenc.c:2869`

Why is it so?

mpromonet, 2017-07-23:
Hi,
This seems related to ffmpeg.
Regards,
Michel.

closed 2017-07-23 by mpromonet