mpromonet/v4l2rtspserver #57: jpeg snapshot

imperialjoy, 2017-05-31:
Hi,

This tool is great for rtsp, good job and thank you!

Is it possible to get a jpeg snapshot of the stream?  What would the URL be?

mpromonet, 2017-06-10:
Hi,
This is not implemented.
For a JPEG capture is could be easy to memory the last picture and give it throught the HTTP server, for instance http://....:8554/snapshot.jpeg
For other stream like H264, this need to store last I-Frame, decompress H264 and recompress in JPEG, more stuff to do.
Best Regards,
Michel.

imperialjoy, 2017-06-10:
It would be a nice feature, since many new cameras do not provide jpeg over http anymore, just h264 over rtsp making them incompatible with 3rd party motion detection, analysis/facial recognition software.

mpromonet, 2017-06-10:
Then it is not only a jpeg snaphot times to times to display some thumbails...

Streaming jpeg over http using multipart seems quite old techonoly, maybe analysis/facial software should decode H264 ?
But this is an other story and the HTTP server of live555 need some improvement to be able to stream HTTP jpeg multipart.

closed 2018-09-27 by mpromonet