mpromonet/v4l2rtspserver #58: Reading the output to v4l2 device with ffmpeg doesn't work

PerryVanKuppeveld, 2017-06-09:
On a Raspberry Pi 3 with a raspicam V2.1 I do the following:
Create 2 v4l2loopback devices (/dev/video2 and /dev/video3)

Converting the h264 stream from the camera to raw with ffmpeg works fine:
`ffmpeg -f v4l2 -input_format h264 -i /dev/video0 -f v4l2 /dev/video3`

Start v4l2rtspserver:
`$ ./v4l2rtspserver -W 1280 -H 960 -Q 100 -O /dev/video2 /dev/video0`

Try converting the /dev/video2 stream to raw with ffmpeg:
`ffmpeg -f v4l2 -input_format h264 -i /dev/video2 -f v4l2 /dev/video3`
ffmpeg starts complaining on non-existing PPS frames

Any idea on the possible problem?

mpromonet, 2017-06-10:
Hi,
This seems the same issue than #56.
SPS & PPS are not added by v4l2rtspServer, then if your device send it, v4l2rtspserver will forward to the output device.
You could enable SPS & PPS repeat using :
 
      v4l2-ctl  -c repeat_sequence_header=1

Otherwise you need to modify output writing moving it from generic part  V4l2DeviceSource to specialized part  H264_V4L2DeviceSource, add marker + SPS + marker + PPS + marker + IDR to the output buffer.

Best Regards,
Michel.

PerryVanKuppeveld, 2017-06-12:
Thanks, changing the setting on the driver dit the trick

closed 2017-06-12 by PerryVanKuppeveld

elipschutz, 2018-10-03:
Hi @mpromonet, I know this is a year old, but can you clarify which method I would need to modify? I'm guessing it's either getNextFrame or extractFrame. And then what is the "marker" byte sequence?
Thanks.

mpromonet, 2018-10-09:
Hi elipschutz,
The problem is how you will get the SPS & PPS, if these are not get from the device read by v4l2rtspserver, you will need to get from an other channel.
If the SPS & PPS are not repeated in the input of ffmpeg command, it will not be present in the ffmpeg output device that is the input device read by v4l2rtspserver (that then never get it)
Best Regards,
Michel.

elipschutz, 2018-10-09:
After spending a couple days working on this it turned out this wasn't actually the problem I was having. Thanks anyway, the server has been really helpful.