mpromonet/v4l2rtspserver #61: CORS - Access-Control-Allow-Origin for HLS and alike

scramble45, 2017-06-22:
The header needs to be set on the server serving the HLS manifests and segments. How can this be accomplished using v4l2rtspserver?

mpromonet, 2017-06-22:
Hi scramble45,
The commit https://github.com/mpromonet/v4l2rtspserver/commit/82ed3868573fe2edfb2800eecc875a89a5f996e6 set the CORS header for HTTP request.
This may helps you ?
Regards,
Michel.

scramble45, 2017-06-23:
Hmm, I’m trying to do a HLS stream It Works in VLC but doesn’t work with any player in the browser keeps saying cross-origin is not turned on.

scramble45, 2017-06-23:
````
XMLHttpRequest cannot load http://192.168.250.26:8554/unicast.m3u8. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://192.168.249.139' is therefore not allowed access.
10:11:06.414 video.js:22963 VIDEOJS: ERROR: (CODE:4 MEDIA_ERR_SRC_NOT_SUPPORTED) The media could not be loaded, either because the server or network failed or because the format is not supported. 
````

scramble45, 2017-06-23:
Thought this was a prior commit, guess I miss the fact that it was new. Going to rebuild and test it. Will close if it works.

mpromonet, 2017-07-30:
Hi scramble45,
Still now I did not found a browser that play the HLS streamed by this streamer, I only success to play using vlc, then the problem is probably not linked to ssl.
Maybe the m3u is too simple for the browser, maybe the http server is too crapy.
This need some deeper investigation.
Best Regards,
Michel.

mpromonet, 2017-09-06:
Hi scramble45,
I submit this fix fro hls.js https://github.com/video-dev/hls.js/pull/1327.
With this HLS is working throught a web browser.
Best Regards,
Michel.

scramble45, 2017-09-06:
Seems to be working well running everything client side, still hoping for HTTPS support in a newer commit. I tried other HLS players in chrome and FF ran into cross origin stuff so just ended up using your example code but tweaked it for just HLS. Here is example of some node.js code  that I whipped up to help me get this going on a Raspberry Pi 3.

````
var express = require('express'),
    logger = require('morgan'),
    moment = require('moment'),
    os = require('os'),
    util = require('util'),
    app = express();

app.use(logger('dev'));
app.use(express.static(__dirname + '/static'));

app.get('/', function (req, res, next) {

        // Basic Authentication
        const auth = {login: 'admin', password: 'yourpasswordhere'}

        const b64auth = (req.headers.authorization || '').split(' ')[1] || ''
        const [login, password] = new Buffer(b64auth, 'base64').toString().split(':');

        // Verify login and password are set and correct
        if (!login || !password || login !== auth.login || password !== auth.password) {
        res.set('WWW-Authenticate', 'Basic realm="nope"');
        res.status(401).send('Password or username was incorrect');
        return
        }

  try {
    res.send(`
      <!DOCTYPE html>
      <html>
                    <link rel="shortcut icon" href="about:blank"/>
                    <body>
                        <h3>HLS</h3>
                        <video controls id="hlsvideo"></video>
                        <script src="hls.js"></script>
                        <script>
                                if (Hls.isSupported()) {
                                        var video = document.getElementById("hlsvideo");
                                        var hls = new Hls();
                                        hls.loadSource('http://192.168.249.229:8554/unicast.m3u8');
                                        //hls.loadSource(streamList[0]+".m3u8");
                                        hls.attachMedia(video);
                                        hls.on(Hls.Events.MANIFEST_PARSED, function() { video.play(); });
                                } else {
                                        document.write("HLS not supported");
                                }
                        </script>
                    </body>
                </html>
    `); // end of html

  } catch (e) {
    next(e)
  }
});

app.listen(process.env.PORT || 80, function () {
  console.log('Listening on http://localhost:' + (process.env.PORT || 80))
});

````

closed 2017-09-06 by scramble45