mpromonet/v4l2rtspserver #62: Periodic file output as per openRTSP?

retrostrobe, 2017-06-24:
The idea is to enable a continuous recording and split the files so there's one movie file per day.

I was wondering if it's possible to do so with the current version or if there's another way?

openRTSP reference:
http://www.live555.com/openRTSP/#periodic-file-output

Thanks

mpromonet, 2017-06-27:
Hi,

This project is an RTSP server, if you should record the stream in splitted file, you could use openRTSP, or ffmpeg ....

Regards,
Michel.

retrostrobe, 2017-06-27:
Thanks for replying. 

Was mainly wondering since there's an option to save as a file, and wanted to find out if splitting was in the pipeline.  

>  -O output: Copy captured frame to a file or a V4L2 device

mpromonet, 2017-06-27:
Hi,

This is clearly not in the list of things I would like to add to this project. I would like to keep it simple and do only streaming. If you like you can use the -O option with a v4l2loopback device and pipeline with ffpmeg to build some splitted files.
But maybe I will changed my mind if I find an easy way to do this....

Best Regards,

Michel.

retrostrobe, 2017-06-27:
Thanks Michel. I really like v4l2rtspserver, and how simple it is.  I'll try the loopback approach for now.

closed 2017-06-27 by retrostrobe