mpromonet/v4l2rtspserver #63: -S segmentation for HLS stream

scramble45, 2017-06-27:
Whenever I create an HLS stream with -S 60 or any other number the stream is no longer accessible outside of using just -S with the default setting. Playing stream using VLC.

Example: -S 5

Result: http://192.168.250.26:8554/5/unicast.m3u8

Stream is not accessible.

````
http error: error: HTTP/1.1 405 Method Not Allowed
http error: error: HTTP/1.1 405 Method Not Allowed
access_mms error: error: HTTP/1.1 405 Method Not Allowed
core error: open of `http://192.168.250.26:8554/5/unicast.m3u8' failed
````

mpromonet, 2017-06-27:
Hi scramble45,

This is because -S could have parameter or not. In such a case, the space character between option and parameter in not accepted. Then, you should use `-S5` or `-S` or `-S60` but not `-S 5` (5 is taken as a capture device).
Regards,
Michel.

scramble45, 2017-06-27:
Ah makes sense thanks

closed 2017-06-28 by scramble45