mpromonet/v4l2rtspserver #65: cmake fails while compiling

ben0110, 2017-06-28:
HI,
I'm trying to run camera server but I'm getting this error while running cmake 
[image]
do you have any idea why i'm getting this error?

mpromonet, 2017-06-28:
Hi,
This is probably because you did not clone the gti repo, but download the source as a zip.
Please try using git clone.
Maybe I will improve the build to insert a dummy version in such a case.
Regards,
Michel.

ben0110, 2017-06-28:
yes it worked with git clone thank you 
best regards,
Amine

ben0110, 2017-06-29:
Hi again Michel,
I was able to set up the server and the driver for the raspicam but still getting an error while starting the server : 
[image]
even though the h264 compressed belongs to the list of the handled formats:
[image]
should i try with the v4l2loopback ? or is it caused by something else ?

Best regards,
Amine

mpromonet, 2017-06-30:
Hi,
By defaultlt v4l2-ctl display device information of /dev/video0, so to be consistent you should try v4l2rtspserver with the same device.
Regards,
Michel.

closed 2017-07-30 by mpromonet