mpromonet/v4l2rtspserver #67: V4L2 options + Pi Zero W settings?

retrostrobe, 2017-07-01:
Hi Michel,

Are you able to explain what these three options do? Is one better than the other for e.g. performance or smoother playback?

> V4L2 options :
> 	 -r       : V4L2 capture using read interface (default use memory mapped buffers)
> 	 -w       : V4L2 capture using write interface (default use memory mapped buffers)
> 	 -s       : V4L2 capture using live555 mainloop (default use a separated reading thread)

Also, I'm using a Pi Zero W and was wondering if you have any recommended settings for streaming over WIFI? I can control FPS, but not the Video bitrate. Any clue how I can control the bitrate?

Thanks.

mpromonet, 2017-07-01:
Hi retrostrobe,

The effect of these options are probably insignifiant. 
Nevertheless -r is for using read api instead of ioctl [VIDIOC_DQBUF](https://linuxtv.org/downloads/legacy/video4linux/API/V4L2_API/spec/rn01re54.html)
The options -w is used only when copying capture to an output buffer (using -O /dev/videoxx) and then use write api instead of mmap ioctl.
The options -s use add capture filedescriptor to live555 reactor then there is only one thread that read the capture device and write to the network. Without there is a queue between the capture thread and the sending thread.

For streaming over WIFI, it is probably better to disable power management, maybe tcp streaming could helps ???  

Best Regards,
Michel.

retrostrobe, 2017-07-02:
Thank you for clarifying the options.

Also, how would you go about limiting the video bitrate so we're not sending too much data over WIFI?

closed 2017-07-05 by retrostrobe