mpromonet/v4l2rtspserver #68: vlc not receive anything from picamera

carlosb1, 2017-07-13:
I would appreciate any idea about this, I don't know how to fix it... I opened a vlc to receive the streaming from a raspberry with a picamera, but It doesn't work and I receive this message:
The installation seemed well installed, and I started the server with this: 

`./v4l2rtspserver  -vv -T`

Also, I activated the kernel module with this: 
`sudo modprobe -v bcm2835-v4l2`

And I receive this message in the raspberry:
```
2017-07-13 15:17:18,420 [INFO  ] - /home/pi/Desktop/fcb_project/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
	level:DEBUG
2017-07-13 15:17:18,429 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:436
	Create V4L2 Source.../dev/video0
VIDIOC_REQBUFS: Inappropriate ioctl for device
2017-07-13 15:17:18,435 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:455
	Create Source .../dev/video0
2017-07-13 15:17:18,437 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:148
	Play this stream using the URL "rtsp://158.109.8.128:8554/unicast"
2017-07-13 15:17:18,441 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
	begin thread
2017-07-13 15:17:18,444 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:206
	V4L2DeviceSource::getNextFrame no data errno:0 Success
2017-07-13 15:17:18,446 [ERROR ] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
	error:Success
2017-07-13 15:17:18,446 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
	end thread
2017-07-13 15:17:52,498 [DEBUG ] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:146
	Queue is empty
2017-07-13 15:18:03,013 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2017-07-13 15:18:03,018 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2017-07-13 15:18:03,027 [DEBUG ] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:146
	Queue is empty
2017-07-13 15:18:13,570 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2017-07-13 15:18:13,575 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames

```

carlosb1, 2017-07-13:
Rebooting the computer, it seems activate the camera it try to do streaming but my vlc not receive anything

```
2017-07-13 15:36:50,361 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:436
	Create V4L2 Source.../dev/video0
2017-07-13 15:36:50,368 [NOTICE] - src/V4l2Device.cpp:115
	driver:bm2835 mmal 85200005
2017-07-13 15:36:50,369 [NOTICE] - src/V4l2Device.cpp:117
	/dev/video0 support capture
2017-07-13 15:36:50,371 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video0 support read/write
2017-07-13 15:36:50,372 [NOTICE] - src/V4l2Device.cpp:120
	/dev/video0 support streaming
2017-07-13 15:36:50,376 [NOTICE] - src/V4l2Device.cpp:178
	/dev/video0:H264 size:640x480 bufferSize:307200
2017-07-13 15:36:50,378 [NOTICE] - src/V4l2Device.cpp:199
	fps:1/30
2017-07-13 15:36:50,379 [NOTICE] - src/V4l2Device.cpp:200
	nbBuffer:1
2017-07-13 15:36:50,384 [NOTICE] - src/V4l2MmapDevice.cpp:70
	Device /dev/video0 nb buffer:10
2017-07-13 15:36:50,847 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:455
	Create Source .../dev/video0
2017-07-13 15:36:50,853 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/main.cpp:148
	Play this stream using the URL "rtsp://158.109.8.128:8554/unicast"
2017-07-13 15:36:50,855 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
	begin thread
2017-07-13 15:36:50,861 [NOTICE] - /home/pi/Desktop/fcb_project/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
	profile-level-id=000001;sprop-parameter-sets=AAAAASdkACisK0BQHtAPEiag,AAAAASjuHyw=
```

mpromonet, 2017-07-13:
Hi carlosb1,
The first thing to try is to check if the streamer works locally, you can do this with openRTSP.
Using openRTSP rtsp://158.109.8.128:8554/unicast, should create a file that you can play with vlc setting demuxer to h264.
Best Regard,
Michel

carlosb1, 2017-07-14:
thanks for your help. I could do it work... I was an error executing cvlc command...
```
sudo modprobe -v bcm2835-v4l2
v4l2rtspserver -W 480 -H 320 -F 30 -T
```
But, the curious is that if I try to read from some computer vision library like python-opencv, It doesn't stream anything (I receive an empty frame). 
```
vcap = cv.VideoCapture("rtsp://158.109.8.128:8554/unicast")

while(1):

    ret, frame = vcap.read()
    cv.imshow('VIDEO', frame)
    cv.waitKey(1)

```
And a last doubt if it is not problem... do you know which parameters are useful to tune the streaming, I receives image with a lot of noise (I was changing the fps, width and height but it is not enough)....

closed 2017-08-31 by mpromonet