mpromonet/v4l2rtspserver #70: Set the camera sensor mode in Raspberry Pi

paol, 2017-07-23:
Is it possible to select a sensor mode (https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes) explicitly when using V4L2? If so, I would like to request that the option be added to v42rtspserver.

This would be equivalent to the -md option in raspivid.

(P.S. thank you for writing this software)

mpromonet, 2017-07-23:
Hi paol,
This software aims to be as standard as possible, then adding specific control for raspberry is not a way I'd like to follow.
You can probably use `v4l2-ctl` to set this option before startting v4l2rtspserver.
Best Regards,
Michel.

paol, 2017-07-24:
> adding specific control for raspberry is not a way I'd like to follow

Fair enough. I'll close this request then.
Do you know whether this operation is possible via v4l2-ctl? I went over the options (there are a lot of them) and couldn't find it.

closed 2017-07-24 by paol

mpromonet, 2017-07-25:
Hi,
I looked to `v4l2-ctl -L` and did not find it, probably it is not supported by the V4L2 driver, it need to look at that.
Regards,
Michel.