mpromonet/v4l2rtspserver #71: uv4l_raspicam died after v4l2rtspserver started

florian-asche, 2017-07-30:
Hi folks,

i am using the latest installed uv4l packages from "deb http://www.linux-projects.org/listing/uv4l_repo/raspbian/ jessie main". I am using latest build by my own v4l2rtspserver.

Hardware is an "raspberry pi zero w" with "pi zero v2 camera"

Starting uv4l_raspicam worked fine, after starting v4l2rtspserver the uv4l_raspicam crashed. This behavior is sine some days. Since the latest update uv4l. I hope we can resolve that issue. For the start i have some logs...

```
Jul 30 16:55:37 cam2 systemd[1]: Starting LSB: Userspace Camera Driver...
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: Starting UV4L Raspberry CSI Camera Driver: uv4l
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [core] Trying driver 'raspicam' from built-in drivers...
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <warning> [core] Driver 'raspicam' not found
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [core] Trying driver 'raspicam' from external plug-in's...
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] Dual Raspicam & TC358743 Video4Linux2 Driver v1.9.53 built Jul 17 2017
Jul 30 16:55:37 cam2 systemd[1]: Started LSB: Userspace Camera Driver.
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] Detected camera imx219
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] Selected format: 1280x720, encoding: h264, H264 Video Compression
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] Framerate max. 15 fps
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] ROI: 0, 0, 1, 1
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <info> [driver] RAW data will be included in the JPEG captures
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [driver] H264 costant bitrate: 17000000
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [core] Device detected!
Jul 30 16:55:37 cam2 uv4l_raspicam[1033]: <notice> [core] Registering device node /dev/video0
Jul 30 16:55:39 cam2 ntpd_intres[805]: parent died before we finished, exiting
```

```
Jul 30 16:56:09 cam2 systemd[1]: Starting RTSP-Server...
Jul 30 16:56:09 cam2 systemd[1]: Started RTSP-Server.
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,551 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:456
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Create V4L2 Source.../dev/video0
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,799 [NOTICE] - src/V4l2Device.cpp:115
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: driver:raspicam 5000001
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,799 [NOTICE] - src/V4l2Device.cpp:117
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: /dev/video0 support capture
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,799 [NOTICE] - src/V4l2Device.cpp:119
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: /dev/video0 support read/write
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,799 [NOTICE] - src/V4l2Device.cpp:120
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: /dev/video0 support streaming
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,818 [ERROR ] - src/V4l2Device.cpp:160
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Cannot set format for device:/dev/video0 Transport endpoint is not connected
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,818 [ERROR ] - src/V4l2Device.cpp:65
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Cannot init device:/dev/video0
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: VIDIOC_STREAMOFF: Bad file descriptor
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: VIDIOC_REQBUFS: Bad file descriptor
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,819 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:502
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Create ALSA Source...default
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,819 [NOTICE] - /root/v4l2rtspserver/inc/ALSACapture.h:64
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Open ALSA device: "default"
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,946 [ERROR ] - /root/v4l2rtspserver/inc/ALSACapture.h:71
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: cannot open audio device: default error:No such file or directory
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,946 [NOTICE] - /root/v4l2rtspserver/inc/ALSACapture.h:120
Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: ALSA device: "default" buffer_size:0 period_size:0 rate:44100

```

mpromonet, 2017-07-30:
Hi florian-asche

This is one problem using closed-source project, you should contact the support.
An alternative is to use the standard v4l2 driver `bcm2835-v4l2` that works fine and if it doesnot works properly you can fix it.

However reading your logs, it seems the setformat fails. 
> Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: 2017-07-30 16:56:09,818 [ERROR ] - src/V4l2Device.cpp:160
> Jul 30 16:56:09 cam2 v4l2rtspserver[1088]: Cannot set format for device:/dev/video0 Transport endpoint is not connected

Then you can start with `-f`(without any parameter), proceeding like this, v4l2rtspserver will not set the format, it will use the current one.

Best Regards,
Michel.

florian-asche, 2017-07-30:
Hi,

just about to send the dev an e-mail, i see that the package got updated today. A h264 problem got fixed in the uv4l-raspicam package.

And yes, closed-source is one big problem ;)

Thanks 
Florian

closed 2017-07-30 by florian-asche