mpromonet/v4l2rtspserver #72: nanopi air issue (armbian)

taegyo, 2017-08-03:
hi, i have a question.
this server only support YUYV source format??
i changed NV12 format, compile and execute is ok. 

but by v4l2loopback dev/video10 and compress_h264(with nv12)
#)./v4l2rtspserver /dev/video10

2017-08-03 16:11:56,519 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:476
        Create Source .../dev/video10
2017-08-03 16:11:56,520 [NOTICE] - /root/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
        begin thread
2017-08-03 16:11:56,521 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:168
        Play this stream using the URL "rtsp://0.0.0.0:8554/unicast"
2017-08-03 16:11:56,640 [NOTICE] - /root/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:81
        profile-level-id=deb6e5;sprop-parameter-sets=h9625c3atrGZ3ratmd62uAAAAAAAAAAwITAJBgUrDgMCGgUABBQAUBzhtmgc4bZ8HOG2jBzhtpwc4bYAAAAAAgAAAAIAAABIHOG2WH3itg8AAABofeK2FAAAAL2b3rblzdq2Da7etgmu3ra4AAAA/a7etv////+5yn/WoPXTQhlthJF2HDu+SLKCmGjCgAAAACgYAAAlAFIXjSmi1YQSnYmaRYIC93dCnAg9gvSKQGa1SSerQ..............
......continue

and connect by vlc. 
server is show this message.
Unable to determine our source address: This computer has an invalid IP address: 0.0.0.0

help me please.

mpromonet, 2017-08-04:
Hi taegyo,
live555, that is used by this porject, select the interface to use sending a multicast packet and reading the interface that receive it, or resolving the hostname.
If you got 0.0.0.0 that means multicast discovery fails and your hostname is not resolvable.
This is usualy good to add your hostname in /etc/hosts (with a real IP, not 127.0.0.1 if you like to reach it from outside).
An other possibility is to use the `-I` option to specify the IP interface to listen to.
Best Regards,
Michel.

TopperBG, 2018-01-19:
@taegyo how do you manage to run it on NanoPi Air, please advice.

TopperBG, 2018-01-19:
I found it 	4l2rtspserver -fNV12 but the help is confusing a little

> -f        : V4L2 capture using current capture format (-W,-H,-F are ignored)
> -fformat  : V4L2 capture using format (-W,-H,-F are used)

TopperBG, 2018-01-19:
But further still no video:
`v4l2rtspserver -fNV12 -I 239.168.15.222
2018-01-19 12:43:45,112 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:548
        Create V4L2 Source.../dev/video0
2018-01-19 12:43:45,203 [NOTICE] - src/V4l2Device.cpp:115
        driver:sunxi-vfe 5000001
2018-01-19 12:43:45,203 [NOTICE] - src/V4l2Device.cpp:117
        /dev/video0 support read/write
2018-01-19 12:43:45,203 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2018-01-19 12:43:45,394 [NOTICE] - src/V4l2Device.cpp:177
        /dev/video0:NV12 size:640x480 bufferSize:0
2018-01-19 12:43:45,394 [NOTICE] - src/V4l2Device.cpp:198
        fps:1/25
2018-01-19 12:43:45,394 [NOTICE] - src/V4l2Device.cpp:199
        nbBuffer:0
2018-01-19 12:43:45,394 [NOTICE] - src/V4l2MmapDevice.cpp:70
        Device /dev/video0 nb buffer:10
2018-01-19 12:43:45,431 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:566
        Create Source .../dev/video0
2018-01-19 12:43:45,434 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:161
        Play this stream using the URL "rtsp://239.168.15.222:8554/unicast"
2018-01-19 12:43:45,434 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
2018-01-19 12:43:45,530 [WARN  ] - src/V4l2MmapDevice.cpp:190
        Device /dev/video0 buffer truncated available:0 needed:460800
2018-01-19 12:43:45,530 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:206
        V4L2DeviceSource::getNextFrame no data errno:0 Success
2018-01-19 12:43:45,531 [ERROR ] - /root/v4l2rtspserver/src/DeviceSource.cpp:107
        error:Success
2018-01-19 12:43:45,531 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:118
        end thread
`

mpromonet, 2018-01-19:
Hi TopperBG,
This project is able to stream a compressed format (like H264, MJPEG, H265) but not YUV like NV12.
In order to compress the stream you can you v4l2tools to compress to an intermediate v4l2 device that could be read by v4l2rtspserver.
You can consult https://github.com/mpromonet/v4l2rtspserver/issues/78 that seems really close to your concern.
Best Regards,
Michel.

TopperBG, 2018-01-22:
Thank you, I've tried very hard with v4l2tools but don't have good results yet.

mpromonet, 2018-01-26:
Hi TopperBG,
This is not related to this project, more to v4l2tool project, it is better to use [issues](https://github.com/mpromonet/v4l2tools/issues)
It seems that the V4L2 ioctl SET_FMT fails.
Did you try some commands like : 
   strace  ./v4l2compress_h264 /dev/video0 /dev/video1

Best Regards,
Michel.

taegyo, 2018-02-12:
as i know, this server use a YUYV format for encoding h.264. 
but, nanopi air armbian v4l2 is not support YUYV, so i changed NV12 format in source and compile.
but do not work. maybe change YUYV to NV12 format encoder. now i'm give up. haha.

mpromonet, 2018-02-12:
Hi taegyo,
You are right, without modifcation v4l2compress_h264 need YUYV, but this is easy to modify.
I made it work on a nanopi air using:
    v4l2convert_YUV -oYUYV --> v4l2compress_h264 --> v4l2rtspserver
Best Regards,
Michel.

closed 2018-09-27 by mpromonet