mpromonet/v4l2rtspserver #78: Can it support USB CAM with YUYV format?

codedigger123, 2017-11-11:
Hi mpomonet:
     Can it support USB CAM with YUYV format?
      I test it with a logitech USB CAM,and play rtsp stream with VLC,but can not play.The log in server side and player side is as following:
   ./v4l2rtspserver -fYUYV 
2017-11-11 16:21:37,941 [NOTICE] - /home/test/v4l2rtspserver/src/main.cpp:548
        Create V4L2 Source.../dev/video0
2017-11-11 16:21:38,055 [NOTICE] - src/V4l2Device.cpp:115
        driver:uvcvideo 84200001
2017-11-11 16:21:38,055 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-11-11 16:21:38,075 [NOTICE] - src/V4l2Device.cpp:177
        /dev/video0:YUYV size:640x480 bufferSize:614400
2017-11-11 16:21:38,095 [NOTICE] - src/V4l2Device.cpp:198
        fps:1/24
2017-11-11 16:21:38,095 [NOTICE] - src/V4l2Device.cpp:199
        nbBuffer:0
2017-11-11 16:21:38,095 [NOTICE] - src/V4l2MmapDevice.cpp:70
        Device /dev/video0 nb buffer:10
2017-11-11 16:21:38,099 [NOTICE] - /home/test/v4l2rtspserver/src/main.cpp:566
        Create Source .../dev/video0
2017-11-11 16:21:38,099 [NOTICE] - /home/test/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
2017-11-11 16:21:38,111 [NOTICE] - /home/test/v4l2rtspserver/src/main.cpp:161
        Play this stream using the URL "rtsp://192.168.116.224:8554/unicast"

while the player side,the exception log:
Sending request: DESCRIBE rtsp://192.168.116.224:8554/unicast RTSP/1.0
CSeq: 3
User-Agent: LibVLC/2.2.2 (LIVE555 Streaming Media v2016.02.09)
Accept: application/sdp

[00007fbc00000958] lua art finder debug: Trying Lua scripts in /home/zds/.local/share/vlc/lua/meta/art
[00007fbc00000958] lua art finder debug: Trying Lua scripts in /usr/lib/vlc/lua/meta/art
Received 101 new bytes of response data.
[00007fbc00000958] lua art finder debug: Trying Lua playlist script /usr/lib/vlc/lua/meta/art/00_musicbrainz.luac
Received a complete DESCRIBE response:
RTSP/1.0 404 File Not Found, Or In Incorrect Format
CSeq: 3
Date: Sat, Nov 11 2017 08:43:04 GMT

[00007fbbfc000e38] live555 demux debug: connection error 404
[00007fbbfc000e38] live555 demux error: Failed to connect with rtsp://192.168.116.224:8554/unicast
[00007fbbfc000e38] core demux debug: no access_demux modules matched
[00007fbc080009b8] core input debug: creating access 'rtsp' location='192.168.116.224:8554/unicast', path='(null)'

mpromonet, 2017-11-11:
Hi codedigger123,
Not directly because v42rtspserver doesnot encode and there is no RTSP/RTP standard for YUYV. 
But you can use an intermediate virtual video driver [v4l2loopback](https://github.com/umlaeute/v4l2loopback) and one encoder from [v4l2tools](https://github.com/mpromonet/v4l2tools) to encode the YUYV to a compress format supported by v4l2rtspserver.
This should works with v4l2compress_h264, v4l2compress_jpeg and v4l2compress_vpx, I have still problem with v4l2compress_x265.
Best Regards,
Michel.

codedigger123, 2017-11-11:
Hi Michel:
    I know.thanks a lots!

mpromonet, 2017-11-11:
Hi codedigger123,
An other way is to include encoding stuff in v4l2rtspserver.
Best Regards,
Michel.

closed 2018-01-14 by mpromonet