mpromonet/v4l2rtspserver #79: Cross compilation issues

roger-, 2017-12-03:
Hi,

I'm trying to cross compile for LEDE (mipsel platform) and I'm getting some errors:

```
v4l2rtspserver/src/main.cpp: In function 'FramedSource* createFramedSource(UsageEnvironment*, int, DeviceInterface*, int, int, bool, bool, MPEG2TransportStreamFromESSource*)':
v4l2rtspserver/src/main.cpp:121:21: error: 'V4L2_PIX_FMT_HEVC' was not declared in this scope
  else if (format == V4L2_PIX_FMT_HEVC)
```

I fixed that by defining `V4L2_PIX_FMT_HEVC` (as some nonsense number) and then I got this error:

```
v4l2rtspserver/src/main.cpp: In function 'int main(int, char**)':
v4l2rtspserver/src/main.cpp:537:40: error: 'basename' was not declared in this scope
     baseUrl = basename(videoDev.c_str());
```

I fixed that by adding `#include <libgen.h>` to main.c and by adding `const_cast<char*>` to the `basename())` calls.

I'm having trouble getting it to link with with log4cpp, but it at least compiles now.

mpromonet, 2017-12-03:
Hi Roger,

I don't understand the problem with V4L2_PIX_FMT_HEVC, it is defined in the submodul https://github.com/mpromonet/libv4l2cpp/blob/5889f878eb4a38e722a600aed6bd9d7a8ce9e1cd/inc/V4l2Device.h. Maybe the submodule is not updated ?

For basename I will, probably, remove it, implementing a simple parsing.

Best Regards,
Michel.

roger-, 2017-12-03:
> Maybe the submodule is not updated ?

You're right, forgot to update the submodule!

I'm having trouble setting the log4cpp directory though and make is failing. I have this custom LEDE toolchain file:

```
SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_PROCESSOR mipsel)
SET(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${CMAKE_SYSTEM_PROCESSOR})
SET(CMAKE_C_COMPILER mipsel-openwrt-linux-gcc)
SET(CMAKE_CXX_COMPILER mipsel-openwrt-linux-g++)

SET(CMAKE_FIND_ROOT_PATH ${STAGING_DIR})

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```

where `STAGING_DIR` is where I have log4cpp installed.

Then `cmake -DCMAKE_TOOLCHAIN_FILE=lede.toolchain` seems to work, but make fails:

```
[ 90%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o
/home/roger/stuff/v4l2rtspserver/src/main.cpp: In function 'int main(int, char**)':
/home/roger/stuff/v4l2rtspserver/src/main.cpp:581:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if (videoCapture->getBufferSize() > OutPacketBuffer::maxSize)
                                         ^
make[2]: *** No rule to make target '/home/roger/stuff/log4cpp', needed by 'v4l2rtspserver'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

Any ideas?

roger-, 2017-12-03:
Turned out that it wasn't reading the toolchain file for some reason and was building for the host instead of cross compiling, so I manually edited the CMakeLists.txt and eventually got it working.

Now it runs on my tiny WT3020 router with a 1280x960 h264 stream from a Logitch webcam, which is pretty impressive. It does crash with memory mapped buffers (the default) as well as the read/write V4L2 capture  interface, but it works with the live555 mainloop (-s option). I believe I experienced this on a Raspberry PI as well). Testing now to see how stable it is.

FYI I used the instructions [in this post](https://github.com/mpromonet/v4l2rtspserver/issues/54) but had to modify a lot of things. I may post some notes here, but ultimately I'd like to make a LEDE/OpenWrt package.

mpromonet, 2017-12-03:
Hi Roger,
Did you have some backtrace of the crash ? 
Best Regards,
Michel.

roger-, 2017-12-03:
Both the default parameters and `-r` give:

```
(gdb) bt
#0  0x0044d474 in V4l2Capture::read(char*, unsigned int) ()
#1  0x0043d470 in V4L2DeviceSource::getNextFrame() ()
#2  0x0043e9f8 in V4L2DeviceSource::thread() ()
```

`-w` actually doesn't crash but gives:

```
2017-12-03 20:00:46,328 [NOTICE] - /home/roger/stuff/v4l2rtspserver/src/main.cpp:552
        Create V4L2 Source.../dev/video0
2017-12-03 20:00:46,330 [NOTICE] - src/V4l2Device.cpp:115
        driver:uvcvideo 84200001
2017-12-03 20:00:46,331 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video0 support streaming
2017-12-03 20:00:46,332 [ERROR ] - src/V4l2Device.cpp:124
        Mandatory capability not available for device:/dev/video0
2017-12-03 20:00:46,336 [ERROR ] - src/V4l2Device.cpp:65
        Cannot init device:/dev/video0
```

mpromonet, 2017-12-03:
Hi Roger,
`-w` is for output. This has no impact on capture, in fact your capture device doesnot support mmap.
Best Regards,
Michel.

mpromonet, 2017-12-15:
Hi Roger,
I remove the basename in commit https://github.com/mpromonet/v4l2rtspserver/commit/6524852e9985397ebec231b04c2b44c011e9e5cd, this should simplify the build on your platform.
Best Regards
Michel

roger-, 2017-12-16:
Thanks Michel, I'll update the LEDE package soon.

BTW I was also experiencing an issue with [this regex](https://github.com/mpromonet/v4l2rtspserver/blob/master/CMakeLists.txt#L16) in CMakeLists.txt. 

`${VERSION}` needs to be `"${VERSION}"` with my version of cmake.

mpromonet, 2017-12-16:
Hi Roger,
This make sens, I modify this in https://github.com/mpromonet/v4l2rtspserver/commit/d61e9cbb5db20ce9b349a8418a1ae8550643699c
Best Regards,
Michel.

roger-, 2018-01-10:
FYI the OpenWrt/LEDE [package was accepted](https://github.com/openwrt/packages/tree/master/multimedia/v4l2rtspserver). I’ll try to update it when you make new releases (you’re welcome to update it too). 

One issue is the binary it produces is rather large (1 MB uncompressed), so if you had ideas on reducing its size then please let me know.

closed 2018-01-10 by roger-