mpromonet/v4l2rtspserver #80: NanoPi Air (Armbian 3.36) can't intit video device

TopperBG, 2017-12-11:
Fully update OS, ffmpeg works with /dev/video0, cap.c works, fswebcam make snaphosts.
`ffmpeg -f v4l2 -channel 0 -s 320x240 -i /dev/video0 -c:v libx264 -preset ultrafast -qp 20 ffmpegtest.mkv`
V4l2RTSPserver compiled and installed w/o errors.
When run command follow errors accures:

> v4l2rtspserver /dev/video0
> 2017-12-11 07:57:51,624 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:548
>         Create V4L2 Source.../dev/video0
> 2017-12-11 07:57:51,716 [NOTICE] - src/V4l2Device.cpp:115
>         driver:sunxi-vfe 5000001
> 2017-12-11 07:57:51,716 [NOTICE] - src/V4l2Device.cpp:117
>         /dev/video0 support read/write
> 2017-12-11 07:57:51,716 [NOTICE] - src/V4l2Device.cpp:119
>         /dev/video0 support streaming
> 2017-12-11 07:57:51,716 [ERROR ] - src/V4l2Device.cpp:159
>         Cannot set format for device:/dev/video0 Invalid argument
> 2017-12-11 07:57:51,741 [ERROR ] - src/V4l2Device.cpp:65
>         Cannot init device:/dev/video0
> VIDIOC_STREAMOFF: Bad file descriptor
> VIDIOC_REQBUFS: Bad file descriptor

TopperBG, 2017-12-11:
BTW there was 2 warnings when compile at NanoPi Air:
`[ 63%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o
/root/v4l2rtspserver/src/DeviceSource.cpp: In member function ‘void V4L2DeviceSource::deliverFrame()’:
/root/v4l2rtspserver/src/DeviceSource.cpp:156:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (frame->m_size > fMaxSize) 
                      ^
`
`[ 81%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o
/root/v4l2rtspserver/src/main.cpp: In function ‘int main(int, char**)’:
/root/v4l2rtspserver/src/main.cpp:581:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if (videoCapture->getBufferSize() > OutPacketBuffer::maxSize)
                                         ^
`

mpromonet, 2017-12-11:
Hi TopperBG,
Clearly the problem is not the build warning.
The error message  indicate that the device doesnot support H264 capture format (which is the default capture format)

> Cannot set format for device:/dev/video0 Invalid argument

In order to make it works on NanoPi, I made some modification in the V4L2 driver https://github.com/armbian/linux/pull/5, use an intermediate [v4l2loopback](https://github.com/umlaeute/v4l2loopback) to compress it with [v4l2compress_h264](https://github.com/mpromonet/v4l2tools) and maybe patch a kernel header that wrongly define some socket option...
Best Regards,
Michel.

TopperBG, 2017-12-12:
Thank you, will look after it and will report back.

TopperBG, 2017-12-12:
BTW something.... or anything going wrong.
V4l2loopback can't be compiled, but installed from apt with v4l2loopback-dkms.
v4l2tools are compiled and isntalled but without v4l2compress_h264.
Even installed kernel source (3.4.113-sun8i) can't find vfe.c to patch ..... :/

TopperBG, 2017-12-12:
Wtih JPEG format looks like service is running but cant be played by VLC.

`v4l2rtspserver -f JPEG -I 192.168.15.222
2017-12-12 06:30:07,134 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:552
        Create V4L2 Source...JPEG
VIDIOC_REQBUFS: Inappropriate ioctl for device
2017-12-12 06:30:07,136 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:570
        Create Source ...JPEG
2017-12-12 06:30:07,136 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:161
        Play this stream using the URL "rtsp://192.168.15.222:8554/unicast"
2017-12-12 06:30:07,136 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
2017-12-12 06:30:07,137 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:206
        V4L2DeviceSource::getNextFrame no data errno:0 Success
2017-12-12 06:30:07,137 [ERROR ] - /root/v4l2rtspserver/src/DeviceSource.cpp:107
        error:Success
2017-12-12 06:30:07,137 [NOTICE] - /root/v4l2rtspserver/src/DeviceSource.cpp:118
        end thread
`

mpromonet, 2017-12-12:
Hi TopperBG,
Take care that '-f' options could take an argument or not, then it should not have space between '-f' and 'JPEG', it should be '-fJPEG'. Giving '-f JPEG' is interpreted as 2 options '-f' means do not set format and 'JPEG' means a device to open.
Anyway this will probably not work, except if vfe driver include JPEG support.
 v4l2compress_h264 is not part of [v4l2loopback](https://github.com/umlaeute/v4l2loopback), it is part 
 of [v4l2tools](https://github.com/mpromonet/v4l2tools), there is also a jpeg compressor.
Best Regards,
Michel.

TopperBG, 2017-12-12:
Yes I've read (as usual) whole the issues reports and bear in mind -fJPEG but that way gives me error again:

> v4l2rtspserver -fJPEG -I 192.168.15.222
> 2017-12-12 07:16:50,035 [NOTICE] - /root/v4l2rtspserver/src/main.cpp:552
>         Create V4L2 Source.../dev/video0
> 2017-12-12 07:16:50,124 [NOTICE] - src/V4l2Device.cpp:117
>         driver:sunxi-vfe 5000001
> 2017-12-12 07:16:50,125 [NOTICE] - src/V4l2Device.cpp:119
>         /dev/video0 support read/write
> 2017-12-12 07:16:50,125 [NOTICE] - src/V4l2Device.cpp:121
>         /dev/video0 support streaming
> 2017-12-12 07:16:50,125 [ERROR ] - src/V4l2Device.cpp:161
>         Cannot set format for device:/dev/video0 Invalid argument
> 2017-12-12 07:16:50,150 [ERROR ] - src/V4l2Device.cpp:67
>         Cannot init device:/dev/video0
> VIDIOC_STREAMOFF: Bad file descriptor
> VIDIOC_REQBUFS: Bad file descriptor

I've read also and nice trick with loopback device but not have a luck again with compile of v4l2tools:

> git submodule update --init h264bitstream
> cd h264bitstream && autoreconf -i -f && ./configure
> configure.ac:26: error: possibly undefined macro: AC_PROG_LD
>       If this token and others are legitimate, please use m4_pattern_allow.
>       See the Autoconf documentation.
> configure.ac:28: error: possibly undefined macro: AC_PROG_LIBTOOL
> autoreconf: /usr/bin/autoconf failed with exit status: 1
> Makefile:132: recipe for target 'h264bitstream/.libs/libh264bitstream.so' failed
> make: *** [h264bitstream/.libs/libh264bitstream.so] Error 1
>

mpromonet, 2017-12-12:
TopperBG 
Then your driver doesnot support JPEG
You should install autoconf, automake ...
Regards,
Michel.

TopperBG, 2017-12-12:
It's a standart OV5640 camera. Works fine with FFMPEG or cap.c (OpenCV) or fswebcam....

TopperBG, 2017-12-12:
It is installed but v4l2tools not compiled:

> apt install autoconf automake
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> autoconf is already the newest version (2.69-9).
> autoconf set to manually installed.
> automake is already the newest version (1:1.15-4ubuntu1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

mpromonet, 2017-12-12:
Did you tried `sudo apt-get install build-essential libtool` (see [README.md](https://github.com/aizvorski/h264bitstream/blob/1f10a107cf7ba5744539a37b98a43783d16de606/README.md) ) ?

TopperBG, 2017-12-13:
Sure, but after re-install libtool (just for sure) I've installed and recommended package libltdl-dev wich also installs libltdl7. Now v4l2tools are compiled !

TopperBG, 2017-12-13:
Shoudn't use nanopi.toolchain when compile it on the ARM bord (NanoPi Air in case) should I ?

mpromonet, 2017-12-16:
nanopi.toolchain is for cross-compiling

closed 2018-01-14 by mpromonet

TopperBG, 2018-01-15:
For the sake of problem - it's Armbian specific, under FriendlyELEC (manufacturer image) it's compiled and works.